prusa3d/Prusa-Firmware-Buddy #3003: [QUESTION] <Files uploaded to directories land in root directory anyway>

hainjedaf, 2023-02-23:
Min+ succesfully upgraded to wifi and connected to PrusaSlicer.

Upon uploading, I choose a subdirectory and filename using / (slashes) as indicated by PrusaSlicer.
Files get placed in USB root.

I've tried:

./dir/file.gcode
/dir/file.gcode
dir/file.gcode

Also I checked if spaces, underscores and minus symbols mattered; they don't.

cb88, 2023-04-04:
I would like this feature to be added also. 

Uploading to directories. And printing from directories. I already used this in the past from the LCD interface so my files were organized and I didn't have to delete anything (since sometimes I need to reprint something again months later).

Prusa-Support, 2023-04-05:
Thanks for the suggestion.
 
 Now please allow me to close this issue as a duplicate of issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2506.
 
Moreover, this behavior is expected to be improved in the near future as mentioned in the relevant issue of the _Prusa Link_ repository https://github.com/prusa3d/Prusa-Link-Web/issues/290.

Michele Moramarco
Prusa Research

closed 2023-04-05 by Prusa-Support