prusa3d/Prusa-Firmware-Buddy #3006: [QUESTION] Input shaping ETA?

gruenehaare, 2023-02-24:
Marlin has implemented Input shaping (albeit experimental) as of 2.1.2. Any idea on when we might see input shaping on the mini? I have some ringing on my prints I'd really like to get rid off.

Prusa-Support, 2023-04-05:
Thanks for your question.

We don't have an estimated time. I'm afraid we don't even have actual plans to implement this at the moment.

he MINI may not be equipped with adequate hardware for advanced features like this.
We can't exclude a "simulated" Input Shaper either but again we have no plans about this and I can't tell if this will be possible to be efficiently implemented in the near future.

The answer to this question is also partially covered at https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3002#issuecomment-1497447993. Please check it out and stay tuned.
If you are satisfied with the answer, please close this issue.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-04-25:
I hope you are satisfied with the answer.

This issue will be closed now but feel free to add further related comments or questions. 

Michele Moramarco
Prusa Research

GithubUser99999999, 2023-09-19:
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3154

Alpha is there now