prusa3d/Prusa-Firmware-Buddy #3007: [ENHANCEMENT] When pausing a print, please park the Z axis higher

ShadowDrakken, 2023-02-24:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MINI/MINI+
**Optional upgrades** - N/A

**Is your suggested improvement related to an existing problem? Please describe.**
Please park the Z axis higher up when pausing a print so that the hot end is accessible. It would help immensely for troubleshooting things like build-up and small clogs if we could get hands and small tools under the hot end.

**Describe the expected functionality**
Ideally, the Z axis should park at 160 when prints are paused to give access to the hot end.

edwardhorsford, 2023-03-04:
+1 for this - I pause to insert magnets, and a higher z would mean I can get to the print more easily.

Prusa-Support, 2023-04-06:
Thanks for your feedback.
Now please allow me to close this issue as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2366. 

Michele Moramarco
Prusa Research

closed 2023-04-06 by Prusa-Support