prusa3d/Prusa-Firmware-Buddy #3009: [BUG] The wifi reconnection not working

Sidexa, 2023-02-26:
**Printer type** - [MINI+]

**Printer firmware version** - [4.4.1+4194]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive 

**Describe the bug**
  With the ESP 01S wifi module as soon as I turn off the mini
and then I turn it back on the wifi is disconnected.

I am systematically obliged to send the file beforehand to the printer with the ssid and password of my wifi.
If I do not turn off the printer, no wifi disconnection to report even with a large STL transfer.

**How to reproduce**
  turn off the printer then turn it back on
**Expected behavior**
 wifi auto reconnect

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

Sidexa, 2023-02-26:
the video of the bug wifi reconnecxion not working
https://drive.google.com/file/d/16UFMf1HKmODj8KDCeJKCjDyjzB2SohiU/view?usp=share_link

Prusa-Support, 2023-04-07:
Thanks for reaching us.
I'm not able to reproduce the problem and we can't exclude ESP module, router, or Wi-Fi signal as being the problem.

If power cycling the printer (OFF > ON) causes the problem, what allows you to reconnect the printer then?
How do you usually solve the problem and restore the connection?

This ESP module is listed multiple times in the [measurements spreadsheet](https://docs.google.com/spreadsheets/d/1ql2TmS8O5Bkuf9SRhfP67RRD1isdThyN96f9M2KdSg4/edit?usp=sharing), but a few times includes negative notes. Please try to reach the manufacturer or the seller for help checking the ESP module correct functionality, or discussing a replacement, and keep us posted.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-04-27:
As you may already know, the printer will attempt a new connection every ~60 seconds.
Please notice that the countdown may not start as soon as the connection gets lost but a few seconds later.
This goes without saying that the connection may fail again for the same reason it failed initially, so a new attempt 60 seconds later doesn't necessarily mean the connection will be re-established.

The reasons for unstable connections vary a lot and may be due to external factors (ESP module quality, Wi-Fi signal power, router settings and protocols...), however, you can also expect natural optimization and improvements on the printer's firmware side over time.
For now, a controlled reproducibility of a problem is strictly necessary for our developers to work on it and similar irreproducible issues should be closed.

This may be a general and irreproducible connection problem like a few more already existing issues.
More specifically, this seems to be a duplicate of issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2772, and therefore, will be closed.

Michele Moramarco
Prusa Research

closed 2023-04-27 by Prusa-Support