prusa3d/Prusa-Firmware-Buddy #3011: [ENHANCEMENT] Please add exFAT support

ShadowDrakken, 2023-02-27:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MINI+
**Optional upgrades** - N/A

**Is your suggested improvement related to an existing problem? Please describe.**
Unable to use certain thumb drives, including those with capacities over 32gb and many newer high speed drives regardless of capacity (Samsung 300MB/s 16gb thumb drive in my case).

**Describe the expected functionality**
FAT32 has been deprecated for years and exFAT is the defacto standard for USB thumb drives and SD cards. Please either add support for, or switch over completely to exFAT to be in-line with modern standards.

Prusa-Support, 2023-04-07:
Thanks for your suggestion.

The request to expand the compatibility to exFAT surely sounds legit but please allow me to list only a few cons (off the top of my mind).

- There are tools to format FAT32 even large flash drives.
- Flash drives directly compatible with FAT32 (≤32GB) are not deprecated but very largely available.
- Mid-quality flash drives ≤32GB are still cheaper than mid-quality larger ones.
- Hundreds of average-size print files are unlikely to fill a merely 8GB large flash drive.
- Extremely large print files are still most likely of negligible size in comparison to a ≤32GB storage, especially for "small" print volumes, and I'm not sure if the printer's CPU would work well with huge files.
- Flash drives may give up for various reasons, especially due to intense use, or if thermally or magnetically stressed. These are not that unlikely conditions in the proximity of a 3D printer. Using the printer's flash drive for large (or long-term) storage purposes may not be advisable.

Abandoning FAT32 shouldn't really be an option a the current time.
This said the issue can stay open, but I'm not sure if this request will be implemented (or can be implemented at all).

Michele Moramarco
Prusa Research

ShadowDrakken, 2023-04-07:
Counter-arguement:

exFAT has lower memory usage and higher performance than FAT32.
exFAT does not require special 3rd party tools to format flash cards as FAT32 does.
exFAT is explicitly designed for Flash devices, helping to extend the lifespan of the flash memory.
exFAT uses scalable cluster sizes instead of fixed cluster sizes, meaning less wasted space (granted, this is a minor issue most people will never see much benefit from)
exFAT supports pre-allocation to prevent fragmentation, which helps extends the life of the flash card.

ZATGamer, 2023-04-10:
@Prusa-Support  your cons are short sighted.

Adding support for another file system expands the capabilities of the printer. Makes it more user friendly. Having a non technical savvy user try to format a 64GB USB Drive they probably have laying around is a frustration for that user. And then they will have a negative experience using your printers.

Another use case you don't even consider are users like me that never delete anything. And with Prusa Link on the Mk4 I want to put 1 USB drive in and forget about it. Only ever using Slicer or Prusa Link to upload files to it. I also have a large collection of production files that I keep on the printers.

Is 64GB ridiculous to put on a printer. Sure. But I got a 64GB short profile stick (SanDisk Ultra Fit) that was less cost for the 16 or 32GB counterparts form a reputable company (SanDisk). And I was frustrated that It didn't just work.

None of your cons that you listed are technical limitations of the hardware. And adding support is good for your user.

Prusa-Support, 2023-04-22:
I see. I think everyone can agree that exFAT has numerous advantages in general.
Sorry, I never mean to say FAT32 is better, but _probably_ most suitable for this use.
I didn't mean that exFAT should _not_ be implemented either. I was trying to find realistic goals for escalating the request, and I'm terribly sorry for the misunderstanding.

I didn't expect it to extend the flash drive lifespan. Maybe that applies when compared to NTFS?
However, if so, that may be one good point. 🙂 

Not sure how soon 32GB could be fully filled up with g-code files in your case, but it would take a relatively long time for average users.
Loading many files and forgetting could also be a point though. 🙂

Lastly, my points clearly are _not technical limitations_.
Whether there are technical limitations or not is unknown to me at the moment. That will be up to the developers.

I hope this clarifies and the issue will stay open for our developers to look into it.

Michele Moramarco
Prusa Research

cb88, 2023-05-01:
Obviously exFAT is better... but given that the USB drive I print off lives with the printer and isn't one I use for other purposes... there is really no reason to implement more than one filesystem as long as that filesystem works on a broad range of operating systems, and FAT satisfies that. 

Adding exFAT increases the maintenance burden on the developers. And would take up extra space on the firmware image, because its a completely separate filesystem from FAT not a simple modification to improve it.

I'd rather they work on must have or annoyance bugs that I have been waiting for them to fix for years... Rather than things I can work around with a simple reformat.

ShadowDrakken, 2023-05-01:
> there is really no reason to implement more than one filesystem as long as that filesystem works on a broad range of operating systems, and FAT satisfies that.

Except FAT32 does NOT satisfy that. Windows holds nearly 60% of the market share, and to use FAT32 on anything but the smallest thumb drives requires third party tools. That's a hassle and can be a risk if a person doesn't know what to get and just grabs some random program from a shareware site that's almost guaranteed to contain malware.

Why force users to hunt down 3rd party utilities? There's no reason to maintain FAT32 at all, drop THAT support and use exFAT INSTEAD.

cb88, 2023-05-01:
Except that isn't true at all. 16GB flash drives are cheap as chips and vastly larger than will ever be needed on current generation 3d printers. Windows 11 formats them as fat32 just fine. If you are buying a larger flash drive you just wasted your money.

Why force developers to develop something that is completely unnecessary. And they ship these printers with USB drives anyway!

Also larger drives can still be formated... with the built in utilities such as drive management snap in, or from the format command line utility, its just the default format menu that doesn't allow this.

ZATGamer, 2023-05-01:
@cb88 your concern is about dev time. I think the devs can decide for themselves what they want to work on. Stop telling us its not needed. I clearly have a need for it and if they want their printers to be truly user friendly its a valid request.

Another idea is to give the Printer the ability to format the card. As long as I can use the full size of my USB Drive I plug into the device, I really don't care what file system it is. I just want to be able to upload from slicer or PrusaLink API and never put the USB Drive into my computer.

And don't bother telling me 64GB is ridiculous, you don't know what I do or how I use the printers.

ShadowDrakken, 2023-05-01:
> 16GB flash drives are cheap as chips and vastly larger than will ever be needed on current generation 3d printers. Windows 11 formats them as fat32 just fine from the format menu. If you are buying a larger flash drive you just wasted your money.

Just because YOU don't need it, doesn't mean others don't. I already had to swap out to a larger one because the 16gb that came with the Mini was full. Some of us use these printers for more than just hobby space.

ShadowDrakken, 2023-05-01:
> The reason to maintain FAT32 is... it works today, requires zero development to continue working, which means developers can work on more important things. At the rate things are going FAT32 will still be around in another 20 years. Also AFAIK exFAT requires licensing to implement (with murky exceptions for OIN / Linux etc... but others likely require licensure)... which is a complete non starter. Implementing it in a commercial product without licensing would open up Prusa to lawsuit from MS.

exFAT is also stable and complete. No further development would be needed there either. And the patents and licensing for it has fully expired.

cb88, 2023-05-01:
> > 16GB flash drives are cheap as chips and vastly larger than will ever be needed on current generation 3d printers. Windows 11 formats them as fat32 just fine from the format menu. If you are buying a larger flash drive you just wasted your money.
> 
> Just because YOU don't need it, doesn't mean others don't. I already had to swap out to a larger one because the 16gb that came with the Mini was full, and I have reasons for keeping the files on the thumb drive. Some of us use these printers for more than just hobby space.

Format your larger drive as FAT32 ... its not difficult at all and does not require 3rd party tools. Fat32 supports up to 2TB still.

ShadowDrakken, 2023-05-01:
> Format your larger drive as FAT32 ... its not difficult at all and does not require 3rd party tools. Fat32 supports up to 2TB still.

Just ignoring that Windows doesn't have a built-in way to do this? Neither the GUI, nor the command line will allow it. Goes back to my previous posts that you're ignoring. You're just arguing to argue at this point.

neophyl, 2023-05-23:
Not arguing either side here as I don't really care but the last post seems to read that windows default wont/cant format a usb drive to Fat32 ?
All the windows machines I have access to (work /home) so ranging from an old Win7 machine to win10 all have options for NTFS/FAT/FAT32 and the windows 10 one I looked at also had exFAT.  That was with the right click>format option.  

In fact FAT32 was the default type for the USB stick I happened to have in the machine at the time.

I would expect any of the operating systems on a full desktop computer to be able to format in any of the file systems..  However I do know that for embedded systems (which we develop where I work) you do have to use licenced libraries to support exFAT.  That's with the STM processor based  boards we use.  I vaguely recall reading ages ago that the Buddy type boards use the same family of processors.

ShadowDrakken, 2023-05-23:
> I would expect any of the operating systems on a full desktop computer to be able to format in any of the file systems.. However I do know that for embedded systems (which we develop where I work) [...]

Windows can read FAT32 of any size just fine, but it can't format FAT32 over 16gb without 3rd party tools.

Licensed libraries are not needed for exFAT, and haven't been for some time now. exFAT has been available in Linux since 2009, with GPL drivers since 2013, and Microsoft officially put exFAT into public domain in 2019. Linux has had exFAT support natively built into the kernel since 2019 as a result. It actually takes effort NOT to support exFAT at this point.

Prusa-Support, 2023-05-30:
Thank you all for your input.
There are surely good insights from all sides, your comments are valuable to us.

The issue will stay open as inspiration for further developments.
Our developers and collaborators will make their considerations about it.

Michele Moramarco
Prusa Research

jmckeown2, 2023-08-06:
From a brief perusal of the sources, It seems like Buddy is getting its FAT32 (and lack of exFAT) support via inclusion of the [TinyUSB](https://docs.tinyusb.org/en/latest/) library.

I find nothing on [their issue tracker](https://github.com/hathach/tinyusb/issues?q=exFAT) to indicate exFAT support is even on their radar. Perhaps filing the Enhancement request with TinyUSB would yield more satisfaction? If THEY add exFAT support upstream, then "pull the latest version" is a much more reasonable request for Prusa Devs.

cplumb, 2023-10-23:
Relevant forum post seems to suggest this might be able to be enabled via a config change in the firmware

https://forum.prusa3d.com/forum/postid/670853/

farchord, 2024-02-06:
Yeah I landed on this myself. I got a new MK4 as-of late last year, and I love it, but I essentially have to scan/repair the USB drive every 2-3 weeks as errors crop up on the partition. Heck, at one point I even had to delete the partition completely and redo it, as even Linux had a hard time figuring out what even was on that partition.

parport0, 2024-03-03:
Does it mean that this is just a matter of setting FF_FS_EXFAT to 1 in include/buddy/ffconf.h?

Prusa-Support, 2024-03-13:
I got confirmation that theoretically, this file system can be enabled on the printer but also that theoretically, this will bring no real benefits to the printer itself - I'm still unsure.

The only technical potential limitation is that larger-size USB flash drives are often not adequately tested on 1.0 USB ports by the manufacturers, and there is a chance that this may affect the data transfer and the flash drive lifespan negatively. However, this is mere speculation.

Our developers will consider testing larger storage USB flash drives and the exFAT file system in the future.

Michele Moramarco
Prusa Research

ShadowDrakken, 2024-03-13:
> and there is a chance that this may affect the data transfer and the flash drive lifespan negatively.

That shouldn't be the case, part of the reason for exFAT is that it extends the life

mitchellj, 2024-06-06:
Came here to add frustration on the lack of support for modern filesystem types. More and more machines are appearing with USB C only support - there are lot of "transitionary" USB Keys on the market that support both USB C and USB A connectors on a single key - however the majority of them are >32 Gig - yes you can still get them but why buy specifically for a single device when you can buy one that can be used for multiple environments not just for PRUSAs lack of (modern) standards support. 

I don't think anyone is saying drop FAT32 support - but adding exFAT support as well instead of adopting the "if it is still kind of working and causing inconvenience lets ignore the issue".

davdpwr, 2025-08-07:
Wow this conversation is so obnoxious. One, there are hella drives </= 32GB on Amazon for under $10. Buy 3 if you really need that much space. You can't make a real good-faith argument that you will fill it up soooo fast that 32GB is horribly inconvenient. 

Should just be argument over at that. Some drives will come in the mail FAT32, some will be exFAT. It's just how it is. If you could find your way to GitHub, you should be able to figure out how to get around the Windows formatter issue. But ok if you insist. Windows 11 actually **_supports_** formatting large drives now. And for Windows 10:

```
diskpart
list disk
select disk 1
clean
create partition primary size=30000
list partition
select partition 1
active
format quick fs=fat32
assign
exit
```

ShadowDrakken, 2025-08-07:
> Wow this conversation is so obnoxious. 

Shame you didn't read it and only read the parts you wanted to complain about.

As stated before, it goes beyond just data capacity. exFAT is more performant, helps extend the drive's life, AND IS ALREADY SUPPORTED BY THE LIBRARY PRUSA IS USING, THEY JUST NEED TO ENABLE IT.

But hey, you get to feel like a big man by shooting people down without bothering yourself to pay attention to the conversation. Congrats.

davdpwr, 2025-08-08:
> Shame you didn't read it and only read the parts you wanted to complain about.
>
> As stated before, it goes beyond just data capacity. exFAT is more performant, helps extend the drive's life, AND IS [...]

No I read this entire thread word for word. Just being real not a "big man." I would hope the fine developers at Prusa have cooler things to do that actually add value to the product, moreso than implementing a second filesystem.

I maintain that you're not really offering anything that's realistically worth the effort, or even the complaint. 

Performance? I don't know your personal use case but i'm sure for 99% of owners, this is never even blip on the radar. Idk how you could hit a point where a tiny bit of time saved is gonna make or break your setup.

Ditto for drive life. Again, they're just so cheap. I don't have the numbers on FAT32/ExFat failure rates, but i don't think that even matters. Either way, USB flash storage has and will always be a consumable item. And your files are most likely "backed up" by way of Prusa Connect storage in the event of data loss. 

And this isn't like, a weird Prusa firmware decision either. Countless industrial machines and higher end hobby machines  - from CNCs and lasers to automated sewing and embroidery, vinyl cutters, etc etc - are the same exact way.

It's never just flip a flag in the library - this software supports an international serially produced product. If they thought it was worthwhile, they'd do it. But not before lots of testing and data collection to ensure it works just as well, doesn't break anything, and isn't an annoying or confusing experience for the 90% of users who will never ever think about it anyway. 

Just my opinion. I Don't think they're "holding out" on us by not supporting ExFat. I think they looked at their giant to do list and came to the same calculation.

bkerler, 2025-08-14:
Quite interesting to see this discussion. The following reflects my opinion and doesn't reflect the opinion Prusa might have on this topic. FAT32 is being used for a reason. It uses less memory and there are already bare metal implementations that are provided from STM themselves. ExFat might seem to be a good choice, but afair using ExFat has to be licensed and most OS do not have support for it out of the box (except Windows). A better option would indeed be to use a journaling filesystem such as ext3/4 (there is no support for that under windows) or NTFS (which also causes a licensing headache) in order to prevent issues with unplugging in a hotplug state, however that will also add up to more RAM and space usage in the firmware. Not saying anything other than FAT32 wouldn't be a better choice ... but FAT32 seems to be the most license-friendly and also is supported by most OS (remember, Prusa is used not just under Windows but also Linux/MAC etc.). Cheers.

farchord, 2025-08-19:
> [@farchord](https://github.com/farchord) ah, good to know that it's available under macos. As for linux, it's not installed out of the box and exfuse needs to be installed before usage (due to licensing as well) and still needs to be installed separately, at least that's the case for ubuntu and debian systems.

I guess it varies from distro to distro. It's enabled out-of-the-box in Fedora.

closed 2025-10-27 by github-actions[bot]