prusa3d/Prusa-Firmware-Buddy #3016: [QUESTION] Table leveling data

eWuPi, 2023-03-02:
**Leave your comments or questions here.**
Please let me know if it is possible to read the table leveling data before printing. They would be useful for mechanical leveling. Maybe via PrusaLink?
--
Proszę o informację, czy jest możliwość odczytania danych z poziomowania stołu przed drukiem. Byłyby przydatne do wypoziomowania mechanicznego. Może poprzez PrusaLink?

bbbenji, 2023-03-18:
Tutaj masz opisane jak odczytać wartości z MLB: https://github.com/bbbenji/PMSBLM#leveling-procedure

eWuPi, 2023-03-18:
Dzięki to też jest dobre. Wymaga jednak zewnętrznych programów. A raka informacja przez www lub na panelu by wystarczyła.

Prusa-Support, 2023-04-27:
Hello, the use of English is strongly recommended for this repository.

Prusa Link won't support this but maybe Prusa Connect, in the future.

You need an external program (e.g. Pronterface) to send the codes `G28` and `G29`.
https://help.prusa3d.com/article/prusa-specific-g-codes_112173

For the regular and correct use of the printer though, all you need would be to ensure that the printer is well squared.
https://help.prusa3d.com/article/squaring-your-mini_158518

Please mind to close the issue if you don't have further questions.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-05-17:
The information on this issue seems to be incomplete so please allow me to close it as invalid.
I hope my answer adequately addressed your question though.

Michele Moramarco
Prusa Research

closed 2023-05-17 by Prusa-Support