prusa3d/Prusa-Firmware-Buddy #3017: Prusa Mini Firmware update issue

juroh, 2023-03-02:
HI there ... I have a problem to upgrade FW fro, 4.1.0  to version 4.4.0. ... I found here bootloader v 2.2. amd new FW 4.4. ... Both files I copy to new usb stick ( 8GB ), insert it to printer and reset it ... and no update start ... Printer start as usual .. 

I tried several usb sticks, several FW version from 4.1.0 to current and printer still start as usual - without any update.

I didnt find any solution of my problem not only here .. 

What I'm doing in wrong way pls ? 

THX A LOT

mix579, 2023-03-02:
Doesn't sound normal. Please check https://help.prusa3d.com/article/firmware-updating-mini-mini_124784 and try the "alternative" method. Ignore that they only mention one not two files at the beginning of the article. The page still hasn't been updated.

Prusa-Support, 2023-04-11:
Thanks for reaching us.

Something may have gone wrong in the middle of the firmware installation.
We have seen this before but the circumstances are not very clear. We are investigating.
However, this is only a temporary firmware corruption and it should be easily fixable.

What seems to be an effective method is the following.

1. Format the USB flash drive (FAT32), then only upload the smaller .bbf file '_Buddy_update_pre_4.4_4.4.1+4194.bbf_' on it.
2. Plug the USB flash drive into the printer, boot it up and the firmware installation should be prompted automatically. 
If nothing happens, "force the installation" (press the **X reset button** once + **wait for 3 seconds** + press the **knob** once), or even try a different USB flash drive.
3. The printer should ask for the second (bigger) .bbf file after. Unplug the USB flash drive, remove the first (small) .bbf file, and put the second (bigger) .bbf file on it. Plug back the USB flash drive into the printer to continue, and "force the installation" if nothing happens.

If this helps and you have no more questions, please consider closing this issue.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-05-02:
Is this issue still relevant?

Michele Moramarco
Prusa Research

Prusa-Support, 2023-05-24:
The issue has been silent so far.
It can be considered stale and will be closed now.

Michele Moramarco
Prusa Research

closed 2023-05-24 by Prusa-Support