prusa3d/Prusa-Firmware-Buddy #3018: [ENHANCEMENT] Filament Unload Heats Bed

1John419, 2023-03-02:
Filament > Unload Filament - For some reason, this command heats the nozzle and the bed. This makes no sense to me. Isn't the nozzle the only thing that needs to be heated to unload filament? Seems like a waste of energy and needless wear and tear on the bed heater. I use this command after the day's last print has finished. The nozzle and bed are both already cool then. Why heat the bed again?

1John419, 2023-03-12:
I have confirmed bed heating is unnecessary for filament unloading. From a cold state, if I manually heat the nozzle to material temperature and then execute Filament>Unload filament, the filament is unloaded immediately without any heating. The firmware obviously only checks nozzle temperature before unloading filament.

Prusa-Support, 2023-04-26:
Thanks for reaching out.

The idea is that heating the bed takes a long time and most often the user un/load the filament in preparation for a print. In other words, the printer will take this chance to possibly get ready to print, cutting the preheat time in case a print will be started shortly after a filament change.

This may not be a strictly necessary behavior though and our developers will consider changing it.
We already have an open issue about this.

This issue will be closed as duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/706.

Michele Moramarco
Prusa Research

closed 2023-04-26 by Prusa-Support