prusa3d/Prusa-Firmware-Buddy #3020: 4.4.1 firmware update kills hotend fan[BUG] <Enter comprehensive title>

wwholden, 2023-03-03:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - 4.4.1

**Original or Custom firmware** - 4.3.4

**Optional upgrades** - Filament Runout Sensor, Bondtech Heatbreak

**USB drive or USB/Octoprint**
USB

**Describe the bug**
Hotend fan does not work during preheat, printing. However it does work when doing a fan check and it does pass the test.

**How to reproduce**
 Heat up the hotend.

**Expected behavior**
Fan to turn on once above 50C

**G-code**
No Gcode needed

**Crash dump file**
No crash dump reported.

**Video**
https://www.dropbox.com/s/tpra0xi7hxer1on/Video%20Mar%2002%202023%2C%208%2013%2002%20PM.mov?dl=0

murk-sy, 2023-03-06:
I've experienced this *once* before on firmware 4.4.0-BETA2, mentioned it in a different issue: #2314 
May be related so I figured I should bring it up. The fan was not physically blocked.

Quotes related to this:
> Additionally, I have managed to produce a condition where the print fan has an error, but the hotend fan stopped spinning while hotend was at 60+°C. This was during a print on one machine (fan 36/255) and during manual control trying to reproduce the error on another. Unfortunately I have found no way to reliably reproduce it yet.

> As I mentioned in the first post, I have had a case where the hotend fan turned off due to this warning - one was during a print (hotend at 245°C!!) and once during reproduction attempts (hotend 60°C). I have not managed to reproduce this yet.

Prusa-Support, 2023-04-22:
Thanks for reaching us.

In the video, I can see the print fan for a few moments, but I can't its blades.
Maybe the blades were moving? Is there a chance that the print fan is spinning instead of the heat-sink fan?

Please carefully review the fans [wiring](https://cdn.help.prusa3d.com/wp-content/uploads/2021/01/Buddy-zapojeni-4.jpg) and report back.

If the issue persists, please check again after downgrading the firmware and/or performing a [factory reset](https://help.prusa3d.com/article/factory-reset-mini_177516).

Michele Moramarco
Prusa Research

closed 2024-06-19 by github-actions[bot]