prusa3d/Prusa-Firmware-Buddy #3022: [BUG] Prusa Mini+ Parking Bug

seifer69er, 2023-03-03:
**Printer type** - [MINI]

**Printer firmware version** - [4.4.1.]

**Original or Custom firmware** - [Original]

**Describe the bug**
This bug is regarding parking which is does not identify the end stop when parking the z axis.
X axis will identify the end stop and works as expected

**How to reproduce**
 This issue occurs repeatedly under the following circumstances.
On initial power on, if the previous print was printed at max height i.e.180mm.
Starting a filament change/load, will force the z axis to go into parking.
The z motor will continually drive and ignore the z axis end stop and continue to spin.
After the z axis is parked, the x axis will test for end stop and move the hotend as expected.

**Expected behavior**
  Expecting the z axis to identify when the top of the z axis limit is reached, similar to auto home function of the printer.

**G-code**
  N/A

mix579, 2023-03-03:
I don't think that's a bug but intended behavior. The Mini doesn't have mechanical end stops, it senses the torque change when idling on the z axis. See https://forum.prusa3d.com/forum/prusaslicer/mini-z-axis-issue-when-parking/. The sound can be disconcerting but shouldn't be a concern.

seifer69er, 2023-03-03:
> I don't think that's a bug but intended behavior. The Mini doesn't have mechanical end stops, it senses the torque change when idling on the z axis. See https://forum.prusa3d.com/forum/prusaslicer/mini-z-axis-issue-when-parking/. The sound can be disconcerting but shouldn't be a concern.

End stop may not the right word, and agree its not a mechanical end stop, However the way the mini detects the limit is not being applied to the Z axis. hence the bug.

ulab, 2023-03-12:
The upper limit for Z is just software based. 

If you turn the printer on or after a reset, the printer doesn't know the Z axis height. After all you could've moved it by turning the lead screw by hand. It only knows Z origin after it did its homing routine which uses the sensor to detect the bed.

After this procedure it knows how far up it can travel.

If I wanted to avoid this issue, I'd probably set the Z position to an "unknown" value and avoid any upward movements unless it has homed all axes. You can't just home automatically after a reset either, because you don't know if there's still something on the bed.

seifer69er, 2023-03-13:
> The upper limit for Z is just software based.
>
> If you turn the printer on or after a reset, the printer doesn't know the Z axis height. After all you could've moved it by turning the lead screw by [...]

Can the Prusa team confirm if this z limit is derived from software, or from firmware? It looks like the x axis is derived from firmware, so the z axis should be also.

ulab, 2023-03-13:
> Can the Prusa team confirm if this z limit is derived from software, or from firmware? It looks like the x axis is derived from firmware, so the z axis should be also.

Firmware is software… :-). End-Switches would be hardware. 

There are no switches on the Mini. It detects the bed (Z=0) using the SuperPINDA sensor and X/Y start and end points sensing motor current. If a motor gets stopped while being powered, the current will rise. There is no detection method for the high Z value. Otherwise the axis would have to go up all the way and back down for every homing cycle.

seifer69er, 2023-03-31:
Please Update the Firmware, the Z motor is capable of detecting z end stop by reading pin 11 of U6. 
[image]
[image]

Prusa-Support, 2023-04-27:
Thanks for reaching out to us.

This is the expected behavior and not concerning in any way.

The X and Y motors can sense homing via collision detection at the end of the axis.
This feature is not used for the Z-axis homing because the PINDA probe has this duty.
The collision sensing is only used on the Z-axis during the XYZ self-test.

Ultimately, detecting a collision in the Z direction is unnecessary and there is nothing to be worried about the current behavior.

Can this issue be closed?

Michele Moramarco
Prusa Research

tobimai, 2023-05-10:
Even if it's not damaging the printer (afaik steppers are fine when loosing steps) it's not necessarily "nice" in my opinion if the axis just runs into the limit for a few seconds. Also it makes a horrible sound.

Prusa-Support, 2023-06-08:
Thanks for clarifying.

Only please be aware that this is not a bug but the expected behavior ever existing even on Prusa i3 printers. Indeed Prusa i3 printers take advantage of that for leveling the X-axis when necessary.

If the noise may pose a problem according to your experience, feel free to keep the issue open, but please consider modifying the title.

Michele Moramarco
Prusa Research

closed 2024-06-20 by github-actions[bot]