prusa3d/Prusa-Firmware-Buddy #3028: [ENHANCEMENT] Allow PrusaConnect Printing when printer is on the reprint screen

markdesouza, 2023-03-09:
Enter what type of printer or upgrade the enhancement applies to.
Prusa Mini+
Prusa Connect

**Is your suggested improvement related to an existing problem? Please describe.**
Currently you can only start a print from PrusaConnect when the printer is on the home screen, which makes sense. However, this means that if a print has finished and you have cleaned the build plate but forget to click home, you can not start a print remotely.

**Describe the expected functionality**
Treat the post print screen (where the options are Home, Reprint etc) the same as the home screen for the purposes of allowing a print to start via PrusaConnect

mix579, 2023-03-09:
Or better, from any screen ☹️

markdesouza, 2023-03-09:
Well I can understand blocking if the printer is currently performing an operation ie printing, un/loading filament etc... but a successful print screen where its sitting idle seems a bit much

Prusa-Support, 2023-04-26:
Thanks for reaching out to us.

This is a known limitation and intentional as stated in the release notes (November 2022).

> *MINI+ must be on the home screen and “print-preview” screen - this is intentional for safety reasons

https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0

This behavior may change or be corrected via new features (e.g. "programmed" statuses via Prusa Connect) in the future.

Michele Moramarco
Prusa Research

closed 2024-06-20 by github-actions[bot]