prusa3d/Prusa-Firmware-Buddy #3030: [BUG] FilamentSensor state and Network settings are not saved on USB drive

maciejbaur, 2023-03-10:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI+

**Printer firmware version** - 4.4.1

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
Original USB drive

**Describe the bug**
When USB drive is plugged FilamentSensor is set to disabled after printer boot. It can be set to enabled and works fine but reset to disabled after restart when USB drive is plugged. FS state is kept only when USB drive is not attached. I don't know if this setting should be saved in some file on USB but supposely yes, like for network settings. It is also not possible to set static IP address via file from USB drive. It is ignored when created manually, and no file is created by the printer software.

**How to reproduce**
Plug USB drive and turn on the printer. FilamentSensor show disabled state after booting. Turn FS ON and restart printer with USB drive plugged. FS will be set to OFF after restart.
Do the same exercise without USB drive and then FS state will be kept.

**Expected behavior**
FS state should be kept (ON/OFF) after printer restart regardless USB drive is plugged or unplugged.

**G-code**
Nothing to do with G-code.

**Crash dump file**
No crash.

**Video**
I can add if really necessary.

ulab, 2023-03-12:
I have just restarted my 4.4.1 Mini by turning it off and on again with a USB drive connected and it did not change the setting from on to off?

maciejbaur, 2023-03-13:
I've tested that several times and also with different USB drive. Always the same behavior. I don't know what to check more. I'll do that again and attach movie.

vinhor, 2023-03-20:
I contacted support, and you have to turn off "FW update", because printer load new FW from USB drive

maciejbaur, 2023-03-20:
Doesn't help. I have FW update disabled but FS is disabled just after booting.

maciejbaur, 2023-03-20:
No .bbf files. The same behavior on other empty USB drive also. Stable situation, when any USB drive is plugged during booting, FS is set to disabled.

HaWiWe, 2023-03-29:
I have got the same problem. Firmware update disabled, sensor settings reset after every reboot.

maciejbaur, 2023-03-29:
Indeed, I doublechecked and this issue happens in combination with network cable plugged. So when I unplugged LAN cable, FS keeps enable state also after rebooting with USB drive plugged.

HaWiWe, 2023-03-29:
Yeah, seems to be it. It's only been happening since i plugged it into the network. Unplugging fixes it, so the bug is connected to the network cable.

maciejbaur, 2023-03-29:
The best new feature in 4.4.1 for me is PrusaLink with possibility to send g-code files via LAN without playing with USB drive. So I hope that new FW will fix FS issue.

ulab, 2023-03-29:
I still can't reproduce it. I have BL 2.0.2, FW 4.4.1 just as you. I'm connected to ethernet and have a USB stick inserted. When I shut off the printer and switch it back on, the FS stays `on`.

HaWiWe, 2023-03-29:
I just tried all the options again: **No USB**/Network: settings stay. USB/**no network**: settings stay. **USB/Network toghter: Filament sensor disabled**. I tried rebooting, shutting it off, disabling and enabling FW update, deleted the update files from the USB, tried a second USB.

gegy1, 2023-04-12:
My filament sensor is also always disabled when i turn of an on again the Mini. :(

greenhalos, 2023-05-29:
I have the same issue.

Prusa-Support, 2023-06-03:
Thank you for reporting and for your precious input.

It is recommended to open separate issues for separate problems/requests.
About the inability of setting a static IP via _Save/Load Settings_, fortunately, we already have an open issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2367.

However, the main topic of the discussion here is clearly that:
_the filament sensor gets automatically disabled **only if** USB drive and Ethernet cable are **both** plugged **while** rebooting_.

Is that accurate?
Has anyone experienced the same problem when the printer is connected to the network via Wi-Fi?

Michele Moramarco
Prusa Research

HaWiWe, 2023-06-03:
Thanks for reaching out, Michelle.

Yes, that is accurate. Both plugged in always disables the sensor on reboot.

I don't have WiFi available, sorry.

maciejbaur, 2023-06-03:
I also don't have wifi on my MINI.

Prusa-Support, 2023-06-20:
Thank you again for your precious input.
At least one reporter in the issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1833 could confirm your thesis.

Because this issue partially duplicates #1833 and #2367, I hope you don't mind me closing it.
Please keep reporting your further findings and comments on the relevant already existing issues.

Michele Moramarco
Prusa Research

closed 2023-06-20 by Prusa-Support