prusa3d/Prusa-Firmware-Buddy #3032: [ENHANCEMENT] Filament Sensor "n/a" if switched off

ulab, 2023-03-12:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - Mini
**Optional upgrades** - Filament Runout Sensor

**Is your suggested improvement related to an existing problem? Please describe.**
I've just spent an hour trying to figure out why my filament sensor did not work and was reported as `n/a` in `Info` / `Sensor info`.

"Not available" meant to me that was - well - not available. So I took the cover  off, checked the connector, cut the zip tie holding the sensor box, took the sensor out of the box, checked the wiring connectivity, put everything back together, fiddled with the textile sleeves for a bit so I could close the lid, …

And then I noticed that it was simply turned off in `Settings`.

**Describe the expected functionality**
Please show the sensor as `off` or `disabled` if it's there, but turned off.

ulab, 2024-05-27:
Hm.

ulab, 2024-08-19:
Can someone verify this is still an issue? I don't have access to my Mini.

danopernis, 2024-08-19:
Seems like the sensor info screen on MINI is not getting properly updated at least since v5.1.0

I am promoting this to bug.

danopernis, 2024-08-19:
~Seems like the sensor info screen on MINI is not getting properly updated at least since v5.1.0~
Actually, scratch that. I double checked it and it is being properly updated.

However, since I already created the ticket, let's try and fix the "disabled" string at least.

Prusa-Support, 2025-01-24:
This should be fixed by now, thanks for your patience.
The new firmware version is available in its almost final release [6.2.0-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-RC1) and includes this fix.
Please give it a go if you don't mind.

Michele Moramarco
Prusa Research

closed 2025-02-13 by Prusa-Support