prusa3d/Prusa-Firmware-Buddy #3036: [BUG] 4.4.1 Issues with mesh bed leveling

Frixon21, 2023-03-17:
**Printer type** - MINI+

**Printer firmware version** - Prusa-Firmware-Buddy 4.4.1+4194

**Original or Custom firmware** - Original

**Optional upgrades** -  Prusa Filament Runout Sensor,CHT nozzle, Octoprint

**USB drive or USB/Octoprint**
  USB/Octoprint

**Describe the bug**
 If i run a mesh bed levelingn the octoprint termial in the return values there's missing results sometimes just a couple point sometimes whole rows: 
example:
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3
Recv:  0 -0.057 -0.116 -0.064 -0.059
Recv:  1 -0.106 -0.036 -0.046 -0.014
Recv:  2 -0.067 +0.+0.003
Recv: 
Recv: X:178.00 Y:176.00 Z:2.00 E:0.00 Count X:17800 Y:17600 Z:820
Recv: ok

**How to reproduce**
 Running G28 followed by G29 in the terminal or using bed visualizer 

**Expected behavior**
 there should 16 clearly defined values 

**G-code**
  G28 G29

**Crash dump file**
  No crash

**Picture** 
[image]

marek26340, 2023-03-30:
Hi, I am having the same exact issue. Wanted to try Octoprint after a long time of just printing from a flash drive, now I am banging my head on the table trying to figure out why am I having so many issues with even just connecting to the printer. Now I'm considering to downgrade the firmware back to 4.3.4.

rinkek, 2023-04-21:
Same here, wonder what this does for actual calibration.

marek26340, 2023-04-21:
Not sure.
In the end I decided to downgrade and ALL of my issues disappeared. You can read up on all of my issues in this issue: #3042

Prusa-Support, 2023-06-03:
Thanks for reaching out to us.

This issue is actually duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2798 and possibly related to other issues mentioned in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2798#issuecomment-1374500595.
I was of the understanding that most recent FW apparently fixed that and a few possibly related issues.

Now I suspect that we are looking at an intermittent glitch. Also some of the reporters in the other issues may have stopped watching the progress of the issue.

I'm considering closing this issue as a duplicate and reopening the old one.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-06-21:
Issue #2798 reopened.
This one will be closed as a duplicate.

Please report further feedback and findings on the other already existing issue.

Michele Moramarco
Prusa Research

closed 2023-06-21 by Prusa-Support