prusa3d/Prusa-Firmware-Buddy #3037: [BUG] Bearing jumping when parking on calibrating

LyxxSK, 2023-03-17:
Hi, i update fw to 4.4.1 and i have problem when calibrating my mini+. 
here is video: https://ulozto.sk/tam/b0c89282-5a2b-43ce-b254-1d68688c8b0f

when i downgrade to 4.4.0 everything is ok..

Spacekoaster, 2023-03-26:
Hello, 

I've got the same exact problem on my Mini+. Firmware 4.4.1.

LyxxSK, 2023-03-26:
I maybe find bug. When i have inserted my usb 32gb in mini+ bug is here. But when i calibrating first layer without inserted usb . Everything is okey. Problem is inserted usb…

LyxxSK, 2023-03-26:
.

migdard, 2023-04-16:
I had the exact same problem some times, but not always. I thought it was one of my belt tension being too high. Have not checked if related to USB key inserted or not, but will try.

Prusa-Support, 2023-06-27:
I'm sorry, we haven't heard clarifications from you and I'm considering closing this issue as invalid and a possible duplicate.

Michele Moramarco
Prusa Research

closed 2023-07-11 by Prusa-Support

migdard, 2023-07-11:
Agree to close.  The issue actually did not reappear for me, after I went trough mechanical calibration in various ways. Thank you.

Prusa-Support, 2023-07-26:
Thank you for your feedback.

Michele Moramarco
Prusa Research

migdard, 2023-09-22:
Hi 

I have had a reoccurrence with the latest alpha firmware, see https://youtu.be/nkIjVXSwUDM 

I had not printed in a while and upgraded to 5.1.0-alpha1 straight away but the problem is totally identical to how it was with 4.4.1.

Basically it happens just before mesh bed levelling, if for some reason the bed is already at one end of the Y axis. Then as we can see it looks like the printer "tries too hard" against what is already max (or min) Y offset.