prusa3d/Prusa-Firmware-Buddy #3039: To long filenames gives a 415 "Unsupported Media Type" "Not a gcode" error

gurgelx, 2023-03-20:
### Description of the bug

I tried to send a sliced model directly to my mini, but I got a 415 response with "Not a gcode" error.
Tried to reslice it and change settings but nothing helped. Then by a chance I changed the name of the file being sent and it worked.

I guess the printer looks for .gcode in the filename and when its to long it just gets cut off. And the printer thinks its an incorrect file type.

(The filename was really long but I usually don't care about the autogenerated filenames, just slice and send to printer)

### Project file & How to reproduce

(The slicer needs to be connected to a physical printer)
* Do any slice,
* Press the "G" button 
* Rename  file to "insta360 tripod mount for 25mm bar v5_insta360 tripod mount for 25mm bar v5_Body1_0.2mm_PETG_MINI_1h36m.gcode"
* Press upload

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.1

### Operating system

Windows 11

### Printer model

Prusa mini

Prusa-Support, 2023-04-27:
Thanks for contacting us.

We are aware that 'space' can cause issues. See https://github.com/prusa3d/Prusa-Link-Web/issues/224.
It would be safe to close this issue as a duplicate.

For now, removing/replacing the 'space' character should fix the problem. Please give it a try.

Michele Moramarco
Prusa Research

machacekch, 2024-08-20:
This is still broken on 6.1.0 on the MK4S. Here's an example file name that triggers a 415 (Unsupported Media Type) HTTP error "Not a GCODE" when uploaded via PrusaLink web interface:

```
file_name_0123456789012345678901234567890123456789012345678901234567890123456789012345678901234.bgcode
```

Note that it contains no spaces or other special characters. Its length is 95.6 and shortening it by just one character to 94.6 makes the upload work. Please try reproducing the issue with this file name and close it only after you've verified that it's fixed (ideally adding a test to make sure it remains fixed).

danopernis, 2024-08-20:
@machacekch thanks for a report. Yes, this is a valid bug report. Status code is clearly incorrect. This happens due to [unchecked filename truncation](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/v6.1.0/lib/WUI/link_content/prusa_link_api_v1.cpp#L182).

Now the difficult part is to decide what the actual status code should be. Perhaps [414 URI Too Long](https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/414)?

machacekch, 2024-08-20:
> @machacekch thanks for a report. Yes, this is a valid bug report. Status code is clearly incorrect. This happens due to [unchecked filename [...]

[403](https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/403) seems reasonable, but most importantly the user should get a good plain text description of what caused it, e.g. "File name too long".

Heliosmaster, 2024-11-24:
Encountered this again today. Thankful for this GitHub issue, otherwise i wouldn't have been able to diagnose it myself.

bithead666, 2024-12-04:
It's kind of silly to leave something like this in the code for so long that people keep crashing into. What would help to make it a higher priority? Anything I can do?

CZDanol, 2024-12-04:
Internal ticket: BFW-6419

zeevo, 2024-12-18:
Encountered this today and thankfully landed on this issue. I must have re-sliced and re-uploaded my file a dozen times 😆

o-vesely, 2025-07-17:
Hey, just a heads up, we agreed on returning a 414 instead. The error code fits more and the 404 was a bug on our side. Will be fixed in the next release.

closed 2025-07-17 by o-vesely