prusa3d/Prusa-Firmware-Buddy #3041: [BUG] <Enter comprehensive title>

Selta, 2023-04-07:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - 4.5.4

**Original or Custom firmware** - Original

**Optional upgrades** - n/a

**USB drive or USB/Octoprint**
USB Drive

**Describe the bug**
During initial self-testing, I got an error about Y-Axis homing:
[image]

The URL on this screen, prusa.io/13305 is invalid (it redirects to https://www.prusa3d.com/). Similarly, the QR code goes to the same URL.
My guess is that URL is generated, based on the error code YAML:
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/lib/Prusa-Error-Codes/13_MK404/error-codes.yaml

Those pages need created - ideally with steps to correct the errors. In this case, re-starting the self-calibration cleared that particular fault.

**How to reproduce**
Was unable to repro. 

**Expected behavior**
Not an error with homing/self-test, but the URL (likely others, as well) need to be valid.

jokohoko, 2023-04-14:
Same issue here! Had a few error codes that have no error page yet. Even a general errors overview page or the documentation pages would be better than sending users to the homepage of prusa shop.

Prusa-Support, 2023-04-28:
Hello, thank you for the report and sorry to see the errors and that the pages were not published yet at the moment the errors occurred - that specific page sadly is not out yet, but here are the first points we'd recommend checking: 

- Make sure there are no obstructions in the path of the Y-axis. For example, there might be a piece of filament stuck around the belt, or on a smooth rod, from a previous print. 

- Check the Y-axis belt tension and adjust it, following our belt adjustment article https://help.prusa3d.com/article/adjusting-belt-tension-mk3-mk3s-mk3s-mk4_112380

- Ensure that one of the motor pulley set screws is aligned with the flat part of the motor shaft. Make sure the other set screw is also tightened, not excessively, and that there is a small space between the pulley and the motor.

Good point on the code not matching the electrical category - reported this to our devs. 

If there are still issues with the printer, do not hesitate to let me know at any time via email, info@prusa3d.com mentioning me, we'll fix the issue - the missing KB article is known. 

Alessandro Pantaleo

Prusa Research

canDry, 2023-05-26:
Same 13350 issue, led to homepage.  Searching on support came up blank.  Re-running the selt-test a couple times seems to have overcome the issue, but out of 4 tries it failed twice.

Prusa-Support, 2023-05-27:
Hello @canDry, thank you for the message and sorry that the error occurred, would it be possible to send me a video of your selftest via email, both if it succeeds or fail, info@prusa3d.com mentioning me? Just to see all movements, we can see if it there might be anything mechanical to resolve, or if there might be a possible firmware aspect to investigate. 

Alessandro Pantaleo
Prusa Research

canDry, 2023-05-27:
> send me a video
Emailed you a video.  I tried another reset/calibration and it passed everything first time... but then the first print had a "collision detected" error on the x/y homing test.  So it's definitely a persistent issue.

canDry, 2023-07-18:
Since updating the Mk4 firmware (currently at 4.7) the issue hasn't happened again.  👍🏼 😎

madito, 2023-08-10:
With my new Original Prusa MK4 that arrived a couple of days ago and version 4.7.2 that error continues to occur.

Selta, 2024-05-27:
I sold off my MK4 so I can't say whether or not this issue is still valid.

Prusa-Support, 2024-05-28:
The URLs and QR codes seem to work just fine, including the one reported in this issue -> prusa.io/13305.
Thanks for checking and confirming.
This issue will be closed now.

Michele Moramarco
Prusa Research