prusa3d/Prusa-Firmware-Buddy #3042: [BUG] Upgrade Firmware 4.4.X not working

1337hium, 2023-04-08:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** 
MINI

**Printer firmware version** 
4.4.1

**Original or Custom firmware** 
Original

**Optional upgrades** 
-

**USB drive or USB/Octoprint**
USB  

**Describe the bug**
When i try to update the Firmware to 4.4.1 the Printer gets stuck at "preparing bootstrap" 

I tried to flash 4.4.0 and it also didn't work (with the support)

I never get a error, it just get stuck

**How to reproduce**
I can downgrade to 4.2.0/4.3.0, and print fine, try the upgrade to 4.4.0 or 4.4.1 and get stuck at "preparing bootstrap" 

**Expected behavior**
The Upgrade to 4.4.X works 

**G-code**
-

**Crash dump file**
-

**Video**
https://youtube.com/shorts/xrY9DMK9UK0

marek26340, 2023-04-19:
Hi. Not exactly related but - I'd recommend you to stay on 4.3.4. The newer features benefit did not outweigh the issues that I have with the latest firmware. For example, the first issue that I got is #3036. Then I was also noticing all sorts of funny behavior, the most apparent issue was that the printer seemed to ignore any acceleration limits, even while homing and mesh bed leveling. The noises seemed like poorly lubed up bearings sticking, but after downgrading back to 4.3.4, all these issues disappeared. Other people also reported issues with their cooling fans stopping mid print for no reason if I remember correctly.

JanTuts, 2023-06-23:
> Then I was also noticing all sorts of funny behavior, the most apparent issue was that the printer seemed to ignore any acceleration limits, even while homing and mesh bed leveling. The noises [...]

This! Ever since upgrading from 4.3.4 to 4.4.1, my MINI moves rather aggressively and sounds like it's about to die...
Considering I don't get to print very often (sometimes a month between prints), I thought I'd save myself the hassle of downgrading again, and just wait out for the next update with a fix, which I though would _surely come soon._

It's been 5 months now, so I guess I'm finally downgrading back to 4.3.4. 
As happy as I am for the existence of the XL and Mk4, I hope the MINI will get some attention again sometime soon-ish (#3097).

Prusa-Support, 2023-07-26:
Thank you for reaching out to us.

Same as https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3017#issuecomment-1503992677, something went wrong during the installation or we are looking at a rare case of firmware/memory corruption butt this should be simply fixable.

Please try the following.

1. Format the USB flash drive ([FAT32](https://help.prusa3d.com/article/sd-cards-and-usb-drives_112291)), then only upload the smaller .bbf file '_Buddy_update_pre_4.4_4.4.1+4194.bbf_' on it.
2. Plug the USB flash drive into the printer, boot it up and the firmware installation should be prompted automatically. 
If nothing happens, "force the installation" (**press the X reset button once** + **wait for 3 seconds** + **press the knob once**).
3. The printer should ask for the second (bigger) .bbf file after. Unplug the USB flash drive, remove the first (small) .bbf file, and put the second (bigger) .bbf file on it. Plug the USB flash drive back into the printer to continue, or "force the installation" if nothing happens.

Alternatively, please try a different USB flash drive and repeat.

In a few cases, a [factory rest](https://help.prusa3d.com/article/factory-reset-mini_177516) before the process may help.

Michele Moramarco
Prusa Research

1337hium, 2023-07-26:
Hi Michele

I tried your steps, but unfortunately same outcome. It now hangs by "preparing bootstrap", a factory reset doesn't change anything :(

Prusa-Support, 2023-09-06:
You should be able to repeat the flashing process by "forcing the installation:
press the X reset button once + wait for 3 seconds + press the knob once.
This time though, please try a different [suitable USB drive](https://help.prusa3d.com/article/sd-cards-and-usb-drives_112291).

We have seen similar reports in the past so bootstrapping was vastly improved accordingly on [FW 4.4.1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1).

We didn't come across cases of major firmware corruption.
Repeating the steps above should do the trick. If not, please get in touch with our [Customer Support](https://help.prusa3d.com/article/customer-support_2287).

Michele Moramarco
Prusa Research

1337hium, 2023-09-22:
Nothing worked, so i tried to install the new 5.1 Alpha and this worked. So my problemis solved

closed 2023-09-22 by 1337hium