prusa3d/Prusa-Firmware-Buddy #3043: Safari PrusaLinlk Authentication with MK4

dsjove, 2023-04-09:
Authentication eventually works in Safari, but login has to be attempted multiple (inconsistent) times.

1) On entering IP Address into address bar 
2) Login appears as expected
3) Enter 'maker' and paste in copied password
4) Press "Log In"
5) Login appears reappears unexpectedly
6) Repeat 2 to 5 times (3)
7) Now Logged in

Maybe related:
Closing tab with authenticated Prusalink sometimes leaves the page authenticated when reopened, sometimes not.
Occasionally I have seen a weird state where printer data is live updating screen but authentication alert or API Alert errors appear.

MacOS 13.2 (a) Beta (22D7750270d)

reiuwu, 2023-04-25:
I have similar issues on iOS 16.4.1 devices. I have not been able to successfully login after multiple attempts.

mrpajzl, 2023-04-28:
Same issue here

greggjaskiewicz, 2023-05-16:
super irritating

Prusa-Support, 2023-06-23:
Thanks for reporting.

I'm not sure if this is the right repository but I see the issue was opened twice, also in the other probably more pertinent repository (https://github.com/prusa3d/Prusa-Link-Web/issues/386). 

This sounds like a problem with Prusa Link, however, we don't seem to be able to reproduce the issue with other printer models. 
We should perform further tests with the MK4 model now and, until then, we can't exclude an issue with the specific MK4 firmware. This issue will stay open for the time being.

Meantime, please save the credentials on your browser after the next successful login.
The browser autofill option should rule out possible user errors at least. We can start from there.

Michele Moramarco
Prusa Research

jvasileff, 2023-09-21:
The cause and solution are describe in #3287

Prusa-Support, 2023-10-18:
Thank you. I will make sure developers are aware of these findings.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-11-25:
Thanks @pyrho for suggesting the Pull Request https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3371.

Michele Moramarco
Prusa Research

closed 2024-06-06 by danopernis