prusa3d/Prusa-Firmware-Buddy #3044: [FEATURE REQUEST] Display total elapsed time after printing is completed on MK4

in1291, 2023-04-11:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MK4]

**Describe the expected functionality**
  The end screen after a print with the MK4 does not display how long did the printer take. This is useful information to know:
[image]

Besides the "100%" it would be great if total elapsed time was displayed e.g "Finished in 1 day, 8 hours, 22 minutes". With maybe additional information like "Started on 22:01 2023-04-04, finished on 06:03 2023-04-05"

Prusa-Support, 2023-05-19:
Thanks for your suggestion.

Michele Moramarco
Prusa Research

Tupson444, 2024-06-04:
Still valid request

danopernis, 2024-06-04:
> Still valid request

@Tupson444 have you tried latest firmware? I am seeing the total print time on the screen after print is completed when using firmware v6.0.1

Tupson444, 2024-06-04:
I did install the latest firmware, but I haven't printed recently, and I haven't paid attention. I just saw that the issue will be automatically locked so I wanted to prevent that, but if it's implemented than it can be closed :)

closed 2024-06-05 by danopernis