prusa3d/Prusa-Firmware-Buddy #3046: [BUG] Print preview on XL is a different model

mix579, 2023-04-14:
**Printer type** - XL

**Printer firmware version** - 4.5.3

**Describe the bug**
Used Prusaslicer to upload (but not print) model A. Then did an "upload and print" of (a different) model B. Print started but the preview on the LCD screen is model A.

murk-sy, 2023-04-14:
This bug also appears on the Mini (but obviously not shown during the print) . I believe the time estimate displayed is also that of the previous model.

May be related to #2686 or #2331 (which I need to reopen).

mix579, 2023-04-14:
#2686 is a very different issue. #2331 may be related but in my case I'm sure it's not a corrupted drive.

HelgeKeck, 2023-04-14:
i can confirm this. display on the MK4 shows only the last print you started manually from the display, never the ones you started with PrusaConnect or PrusaLink

Prusa-Support, 2023-06-02:
Thanks for reporting.

The issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3057 actually duplicates this one so I added both printer model labels (`XL`+`MK4`) here and we will consider closing the other issue.

Have you had the chance to upgrade the firmware and test again with [FW 4.6.1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.6.1)?

Michele Moramarco
Prusa Research

Prusa-Support, 2023-08-22:
I don't seem to be able to reproduce this issue anymore.
It seems safe to say that the issue is gone with the more recent firmware releases.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-09-13:
This issue has apparently been solved over the last few firmware releases and is stale too.

Michele Moramarco
Prusa Research

closed 2023-09-13 by Prusa-Support