prusa3d/Prusa-Firmware-Buddy #3047: [BUG] Crash when setting up wifi credentials

jokohoko, 2023-04-14:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - 4.5.4

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint** - Original USB drive

**Describe the bug**
I followed the interface to create credentials file in order to connect to wifi. The interface showed a line saying it was working on it, and after about 20seconds crashed showing a full red screen with watchdog timeout. After a restart doing the same thing it says file has already been created. To be sure i told it to make it again and this time it did work. 

**How to reproduce**
No clue

**Expected behavior**
No crash, just a file on the USB drive

**Crash dump file**
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/11234588/dump_buddy.zip)
(github does not support this file extension .bin > Would it be possible to give it an extension that is supported?, this time i zipped it.)

Prusa-Support, 2023-06-14:
Thanks for reporting.

Do you happen to have the `prusa_printer_settings.ini` configuration file that was being loaded when the issue occurred?
Do you happen to have a picture of the watchdog screen?
Do you happen to remember if there was a specific error message on the screen?

If the problem is not reproducible we may not confirm a bug and it could be hard for our developers to work on it.
Also, there have not been similar reports so far. 
Has this issue ever occurred again?

Michele Moramarco
Prusa Research

Prusa-Support, 2023-07-04:
We may likely be looking at a temporary glitch.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-07-04:
We may be looking at a temporary glitch.
Moreover, there have been a few firmware upgrades so far which may have addressed a few possible related problems.

I suggest closing this issue if the problem doesn't persist, can't be documented, or can't be reproduced consistently.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-08-08:
We can't reproduce the issue and the dump file was not sufficient to confirm a bug.

It might be the case of a temporary glitch.
The last few months' updates made the firmware more robust, reducing chances for similar problems to happen again.
 
The issue will be closed now.
Would the issue return, please document it with further dump files and evidence. 

Michele Moramarco
Prusa Research

closed 2023-08-08 by Prusa-Support