prusa3d/Prusa-Firmware-Buddy #3048: Progress Bar Suggestion

IanPendlebury, 2023-04-15:


gold1640, 2023-04-19:
Mk4 firmware also does not show the time to a colour change - would be nice to add this to the main graphic sreen, perhaps as well as current z hight, nozzle and heatbed temperatures.

MrJones4u, 2023-05-01:
Yes, time and layer count till next color change sounds like a great idea. 
Also I find the current Z height is apart from the remaining time the one most looked at values.

Prusa-Support, 2023-06-01:
Thanks for your suggestion.

However, this issue is pretty much a duplicate of the existing issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1062.
Please allow me to close this issue as a duplicate and feel free to share your further ideas and suggestions on the already existing issue.

Michele Moramarco
Prusa Research

closed 2023-06-01 by Prusa-Support