prusa3d/Prusa-Firmware-Buddy #3049: MK4 PrusaLink extremely slow upload over Wi-Fi

H0nzik1, 2023-04-16:
**Printer type**
MK4

**Printer firmware version** 
4.5.4

**Original or Custom firmware** - [Original/Custom]
Original 

**Optional upgrades** 
Nothing

**USB drive or USB/Octoprint**
 USB drive (PrusaLink)

**Describe the bug**
Upload speed over PrusaSlicer or Web interface to the MK4 printer is like 0.1MB/s max. So 10MB file takes forever to upload. MK3S+ with RPi Zero with PrusaLink was like 10x faster.

**How to reproduce**
  Just upload file to MK4 printer. My Wi-Fi router Asus AX68U is 2 meters from printer only.

**Expected behavior**
I expect at least 1MB/s upload speed like with MK3S+ with RPi Zero

**G-code**
  Any
**Crash dump file**
  does not exist any

**Video**

arekm, 2023-04-16:
(same for wired ethernet - 0.3MB/s)

H0nzik1, 2023-04-16:
> (same for wired ethernet - 0.3MB/s)

OMG. I have tried also different USB drive which can do 100MB/s write speed, but it did not helped either.

arekm, 2023-04-17:
For wifi:

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/e688bdf53b0296267a67c31f21960c937193d89d/lib/WUI/espif.cpp#L71-L72
comment suggests that maximal possible wifi speed is about 550KB/s (4.5Mbps) and it's not done because of some corruption.

For wired ethernet the driver copies between buffers:
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/e688bdf53b0296267a67c31f21960c937193d89d/lib/WUI/ethernetif.c#L241
which means it is not a zero-copy driver (like modernly written network drivers are, even for STM32 - https://github.com/lsgunth/lwip_contrib). Copying buffers will always be slow but no idea if that's the only reason for bad 300KB/s wired ethernet speed.

Fail-Fast-V0, 2023-04-20:
My upload speeds are unusable. I noticed the same problem with my Prusa Mini (also running a buddy board).

H0nzik1, 2023-04-26:
Downloading gcode via browser is also insanely slow.
[image]

mrlinuxfish, 2023-04-30:
I'm also having very slow uploads to prusa link on my MINI+. I think this may be a similar/ related issue. Uploading by browser is also very slow compared to uploading to the old solution which was a rpi3 running octoprint

levivoelz, 2023-05-15:
Same. Uploads slow for me in Slicer then sometimes I get a 408 timeout error. I haven't tried in the browser but I assume it's the same problem.

n4bft, 2023-07-01:
+1, this is an unusable level of slow. I've begun using a combination of a pi running usb mass storage gadget (to remotely load files on/off) and use prusa link to control the printer and start/stop prints.

jgreely, 2023-07-04:
What is the *intended* performance of the ethernet and wifi for uploading print jobs? If it's not "multiple megabits per second", then it's just a checkbox marketing feature of no value to end users.

FYI, my 3-year-old Dremel 3D45 has lightning-fast uploads. I was about to buy a Mk4 to replace it until I found out I'd have to print via USB.

abjugard, 2023-07-15:
Can confirm I'm also experiencing this. Fast USB-drive (I assumed this was the issue initially as the drive shipped with my MK4 was INSANELY slow), slow upload speeds both over Wi-Fi and Ethernet. Very irritating when uploading large gcodes. There has to be something we can do about this @Prusa-Support.

neuron-whisperer, 2023-07-24:
+1

Transferring files to and from the MK4 (firmware 4.7.0) is often very slow - on the order of 100kb/second or less.

In general, slow transfer speeds are merely an annoyance, but the issue becomes more significant when transfers of larger files (> 20mb) result in a timeout. On several occasions, I have had to retry file transfers due to timeout issues. For one model that involved a 20mb file, repeated attempts failed, and I needed to retrieve the flash drive and copy the file manually.

The issues also occur for downloads of files stored on the printer - six attempts to download a 20mb file have all failed with timeout messages.

The odd part of this behavior is that the web interface is generally very snappy and responsive; it does not show the same dialup-quality connection issues as file transfer. Only file transfer operations appear to exhibit these performance issues.

abjugard, 2023-08-13:
Same behaviour on 5.0.0 RC.

bobasaurus, 2023-08-14:
I noticed something odd.  WiFi prusalink file transfer speeds were pretty good in 5.0.0-alpha4, but when I upgraded to 5.0.0-RC it went back to being slow again.  I tried updating the WiFi module software from the networking menu on the printer as well, it didn't make any difference.  I think there was some regression of this issue between alpha4 and RC.

drchslrk, 2023-08-15:
> I noticed something odd. WiFi prusalink file transfer speeds were pretty good in 5.0.0-alpha4, but when I upgraded to 5.0.0-RC it went back to being slow again. I tried updating the WiFi module [...]

Same here!

frozenfoxx, 2023-08-15:
Same problem here, it's absurdly slow

devbyrd, 2023-08-21:
Same. Very disappointed with this. Alpha/Beta or not, sub 1Mb isn't acceptable. Any thoughts from a prusa dev?

MartinMajewski, 2023-08-21:
Uuurgh! I encountered this issue with my new MK4 a Minute ago. This is not usable in its current state. I will stick to OctoPrint. Very disappointing.

bobasaurus, 2023-08-21:
> 

Speeds were pretty good on alpha4, but they're back to being slow again on the latest RC.

santiac89, 2023-08-24:
Just discovered this after finishing upgrading my MK3S. Is there any update on it?

bca3244, 2023-08-26:
Bump... these upload speeds are extremely frustating.

abjugard, 2023-08-27:
@josefprusa Can we get an official response on this issue? FYI, it's not just WiFi, it's Ethernet too.

TeTeHacko, 2023-09-06:
no progress ? just have my MK4 and have same problem on Ethernet, speed is werry slow uploading od 11MB file is pain

StephanCassel, 2023-09-07:
@josefprusa We were just about to order three Prusa MK4 but have put the order on hold until the matter is clarified by Prusa. Was hoping to hear more from Prusa. Silence is not a good sign. We are very happy with Prusa products in general and hope it will be resolved soon. There are alternatives.

TravisWilder, 2023-09-08:
> @josefprusa We were just about to order three Prusa MK4 but have put the order on hold until the matter is clarified by Prusa. Was hoping to hear more from Prusa. Silence is not a good sign. We are very happy with Prusa products in general and hope it will be resolved soon. There are alternatives.

There is no Silence
Josef as mention it last week and that it will be address - the new version should arrive this week

StephanCassel, 2023-09-08:
Excellent news. Thanks 😊

CodePhase, 2023-10-15:
I am running 5.0.0 and still have this problem.  On my switch, the speed reports to be 100M _Half Duplex_

machinekoder, 2023-11-15:
> There is no Silence Josef as mention it last week and that it will be address - the new version should arrive this week

I'm still seeing this problem with 5.0.1. I assume this was not the mentioned fix version? A lot of times the slowdown is followed with 408 HTTP error. Wifi or Ethernet doesn't matter.

MartinMajewski, 2023-11-15:
> > > @josefprusa We were just about to order three Prusa MK4 but have put the order on hold until the matter is clarified by Prusa. Was hoping to hear more from Prusa. Silence is not a good sign. We are very happy with Prusa products in general and hope it will be resolved soon. There are alternatives.
> > 
> > 
> > There is no Silence Josef as mention it last week and that it will be address - the new version should arrive this week
> 
> I'm still seeing this problem with 5.0.1. I assume this was not the mentioned fix version? A lot of times the slowdown is followed with 408 HTTP error. Wifi or Ethernet doesn't matter.
> 
> The interesting thing about this is that it worked fine the first day or so after the new printer arrived. It only got worse over time.
> 
> EDIT: I managed to fix the problem in my case: I used the printer with Ethernet in the beginning. I moved it to another room after some time because of noise and switched to wifi. For convenience reasons I used the same static IP address for both interfaces, the main reason I chose to do that was because I thought only 1 interface is active at a time. Apparently this caused the problems. I've switched Wifi back to DHCP and left Ethernet with the static address, upload works again and also Prusa Connect, which also stopped working.

Even with different IP addresses, I have some connection breakdowns now and then. Most times, it is related to the utterly unintuitive behavior of PrusaLink, which considers the printer's status (like being in a filament-changing process while the file gets uploaded) but does not tell you what exactly caused the issue (in PrusaSlicer). Sometimes, the file has been uploaded successfully, and I still get an error response, or it cannot overwrite an existing file, which happens when you make only minor changes to the slicing settings that do not affect print time and generate the same filename.

It is a mess, and slow transfer speeds are only a part of it. I guess the "introduction of the innovative binary gcode format" is a PR stunt to work around the hardware limitations of the entire network interface.

TeTeHacko, 2023-11-15:
yes bgcode and G-code streaming, new features, new bugs - https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3475 its only ugly workaround of the real problem ...

Dilbertus, 2023-11-27:
Testing my Mini and MK4 - sadly this more and more looks like a bad joke - Where has all my money gone buying Prusa printers? Such a bad job of implementing all the network stuff, PrusaLink and connecting to Wifi - so embarrassing.
I would be fired in my job (and even would fire anybody) delivering such a bad work. The PrusaLink feature a ... (bad) proof of concept... - not any more. Downloading shows 100% after a second - than waiting for minutes (Ethernet) without any progress bar.... Now is 6 MONTH later, other companies had constructed new printers with complete eco system in this time. 
bgcode is a awkward workaround - not a solution at all. Who ever is responsible for the software (excepted PrusaSlicer) should look for an easier job. This is imho no longer acceptable.
Programming a web interface isn't rocket science - What am I missing?

samuelT2, 2023-11-27:
@Dilbertus Luckily it's your "imho" - Prusa stands out as one of the rare entities in the field where all the code is open source. If you have innovative ideas or believe you can contribute improvements, don't hesitate to create a pull request. While there might be some issues with over-the-air (OTA) uploads, personally, I find it to be a negligible concern, given the efficiency and ease of use of the entire ecosystem. I'm confident that a multitude of individuals are well aware of any existing problems and are actively working towards enhancing the situation.

Regarding Bgcode, some might perceive it as a somewhat awkward workaround. Typically, people prefer utilizing less space, but if that's not your preference, you can simply disable it. And please, spare me the "but I paid money and I want the full whatever" argument - conduct your research before making a purchase. If OTA bandwidth is a critical factor for you, perhaps consider exploring the latest "printers with a complete ecosystem" that have been developed within 6 months.

TeTeHacko, 2023-11-27:
> @Dilbertus Luckily it's your "imho" - Prusa stands out as one of the rare entities in the field where all the code is open source. If you have innovative ideas or believe you can contribute [...]

So put it this shitty slow speed of all ethernet inerfaces to printer specification. I expect more than few Mbps on "LAN" interface in these days, if i know this before i will probably buy another printer but i stupidly thinks that 100Mbps interface or modern wifi works as usual not as modem line before 20 years

abjugard, 2023-11-27:
> @Dilbertus Luckily it's your "imho" - Prusa stands out as one of the rare entities in the field where all the code is open source. If you have innovative ideas or believe you can contribute [...]

Being open-source is great, but it doesn't excuse shitty implementations, which this absolutely is.

I wouldn't go so far as to say I would have bought another printer solely because of this. But they are advertising PrusaLink/PrusaConnect as features, and the transfer speed at present is not anything anyone in their right mind would be expecting from a high-end device in 2023.

If the problem is ultimately a slow bus to the display-mounted USB drive and not solvable via code, then allow us to use the USB-C port for storage if we want, those relying on network transfers won't need easy access to the USB drive anyway.

vindolin, 2023-11-27:
> To counterbalance this negativity. Since the firmware update (+ wifi firmware update), I've been satisfied with the time it takes to upload models to the printer.

O RLY? How big are your models?
Bigger models e.g. with higher infill and gyroid pattern take FOREVER to upload using wifi or ethernet, even as binary files.
My mk3s with a flashair card runs circles around a mk4 no matter what file type.

abjugard, 2023-11-27:
> > To counterbalance this negativity. Since the firmware update (+ wifi firmware update), I've been satisfied with the time it takes to upload models to the printer.
> 
> O RLY? How big are your models? Complex, bigger models e.g. with higher infill and gyroid pattern take FOREVER to upload using wifi or ethernet, even as binary files. My mk3s with a flashair card runs circles around a mk4 no matter what file type.

Which is totally bonkers. A FlashAir card is a damn €40 SD card that besides implementing storage, also contains a WiFi implementation that beats that of a €1199 top-end 3D printer. That's insane and completely unbelievable.

samuelT2, 2023-11-27:
As a noteworthy addition, I'd like to highlight that with the latest firmware update (5.1.0), my printing process now commences promptly, thanks to the newly integrated streaming feature (independently of the file size and g-code format):

> With G-code streaming support, you can start the print without downloading the entire file to the printer beforehand. Instead, only a small portion of the file is downloaded, allowing the print job to start almost immediately. The remaining G-code is automatically downloaded while the print is in progress. In most cases, the print starts within seconds.

(from [the release 5.1.0 notes](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0))

vindolin, 2023-11-28:
>
Ok, that's a clever workaround if it works, I'll give them that.

neuron-whisperer, 2023-11-28:
The last thing in the world that 3D printing needs is yet another failure mode. What happens if you start streaming the model, your printer starts printing it, and then you have a network failure that interrupts the stream?

I will never use this feature. I will always want to see that my model is 100% uploaded, stored, and preprocessed before hitting the Print button.

abjugard, 2023-11-29:
> I will always want to see that my model is 100% uploaded, stored, and preprocessed before hitting the Print button.

Corruption free, which the current networking stack doesn't seem to promise.

machinekoder, 2024-01-22:
> So almost a year later and nothing has changed. Have Prusa no plans of fixing this?

Well, the current state may not satisfy all users, but saying that nothing has changed is also not correct.

TeTeHacko, 2024-01-22:
bgcode is only ugly workaround of the real problem and i didnt see any change here and please dont mention gcode streaming as another solution/fix for this problem until it works on local network and not on cloud only and streaming is of course another workaround not fix of real problem ...

abjugard, 2024-01-22:
It should be noted that since print times where the files are large are generally measured in hours, an upload taking a few minutes might not matter to a lot of people, and I'll agree with the people saying that there are/have been more serious issues to address, which I'm sure is how Prusa has prioritised internally as well.

All I personally need is an acknowledgement that this issue isn't forgotten, is deemed fixable, and will eventually get prioritised once the XL, MK4 and MMU3 happy-case flows are stable and all corruption issues are ironed out.

erwan, 2024-01-22:
> It should be noted that since print times where the files are large are generally measured in hours, an upload taking a few minutes might not matter to a lot of people, and I'll agree with the [...]

Many people still want to monitor the few first layers of their prints, so in that case you still need to wait.

And yes, it's not the most serious issue out there but it's frustrating because it's hard to understand how network can be that slow in 2024... I don't know of any device released in the last 10 years that has such a slow network stack.

abjugard, 2024-01-22:
> Many people still want to monitor the few first layers of their prints, so in that case you still need to wait.
>
> And yes, it's not the most serious issue out there but it's frustrating because it's [...]

Completely agree on all your points.

Prusa-Support, 2024-01-23:
Thank you all for your feedback.

I seem to understand this issue is based on the initial firmware from April 2023, because, originally, the transfer speed was significantly slower. It was improved shortly after with [FW 4.6.1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.6.1) (Apr 29, 2023), and further with [FW 5.1.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0). Rest assured, we will perpetually improve the firmware over time though.

Surely there is always room to improve the transfer speed and I understand not everyone appreciates binary G-codes, but please notice data transfer and USB writing were actually improved over time, and the .bgcode format can be disabled if the user wants.
The G-code streaming is rather safe too as the print will pause in case of network failure, and can simply be resumed after (see [release notes](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0)). However, even mildly detailed/large print - say slightly above 20MB - would probably fully upload during the print preparation, leaving adequate time for the user to monitor the process and the correct start of the print at once.

I wonder if we - or the author - could consider closing this issue which is populated with scattered problems and requests. We could close this one _in favor of other more recent and specific issues_.
A separate listing of different issues would help us understand which of the mentioned problems is yet to be fixed, what aspect of the firmware is most concerning for the users, and where our developers should put their largest effort next since the development plans are still quite packed.

Michele Moramarco
Prusa Research

abjugard, 2024-01-24:
20MB should upload in seconds. When it does this issue could be closed.

t-makaro, 2024-01-25:
G-code streaming isn't a solution until it works without the cloud, and so large files are still way too slow to upload with PrusaLink. (Wifi or Ethernet). Really, the upload speed should be about 10Mb/s at least for this issue to be closed IMHO.

Dilbertus, 2024-01-25:
Looking into it: (Hardware/Software) - I think we need an honest answer now - after waiting so very, very long. The options for improving speed are limited because someone chose to use inadequate (cheap) networking hardware - and in particular a simple, slow interface to the processor.
What can you expect? Workaround with binary code, because software can't overcome bad design decisions? - And the hope that some developer will come up with a brilliant idea.
Again: Please give an honest answer – not “we should close”. Am I wrong?
Dilbert

abjugard, 2024-01-26:
> Please give an honest answer – not “we should close”. Am I wrong? Dilbert

No this summarises exactly how I feel.

pwojewodzic, 2024-04-14:
Same here... very slow uploading speed - both WiFi / LAN connection.

Prusa-Support, 2024-04-20:
The transfer speed was undesirably slower than expected one year ago, when Prusa MK4 was released and this issue was opened.
The transfer speed was improved within a few weeks, virtually solving _this specific issue_.
Requests and reports after FW 4.6.1 (Apr 29, 2023) should theoretically be posted as a separate issue.

Connection stability, transfer speed, and writing speeds have been improved in various ways over the last year, bringing their parameters somewhere close enough to the hardware limits.
I strongly suggest closing this issue, bearing in mind that: this is a device of limited data usage; external factors like router & distance can also play a role (_only_ to some extent); further requests should ideally be listed separately.

I'm positive that we will see further improvements in these regards, however, more ideas and external contributions for further firmware improvements will always be welcome in the form of new separate issues - as specific, realistic, and concise as they can be -  or Pull Requests - especially.

Michele Moramarco
Prusa Research

abjugard, 2024-04-20:
> The transfer speed was undesirably slower than expected one year ago, when Prusa MK4 was released and this issue was opened. The transfer speed was improved within a few weeks, virtually solving [...]

Should we open a separate issue for Ethernet then? Or can you be more specific what you consider to be "close enough to the hardware limits", because uploads of a few mb over ethernet still take tens of seconds when they should be instant.

miltieIV2, 2024-05-09:
Firmware 6.0.0+14794
Switched from WiFI to Ethernet - no apparent improvement

I'm seeing 22 minute PrusaLink upload times for a 131356101/(22*60) = 99512 bytes pers second
PrusaSlicer to PrusaConnect gave a weird offline error.

Dilbertus, 2024-06-30:
Hello!
I was kind of amused "close enough to the hardware limits", lol.
Yes, that is true, but because you have chosen (as i wrote before): 
> inadequate (cheap) networking hardware - and in particular a simple, slow interface to the processor. >
Could this circumstance be the culprit?
One could see this "close enough to hardware limits" as far an admission as a prusa employee is allowed or willing to do.
Anything wrong with this conclusion?

Yes, with this chosen hardware.. no more improvement is possible - but isn't prusa otherwise well known to offer updates?  What makes things worse for me is that other printers in my farm (for a fraction of price) do the transfer fast, easy and effortless... As the saying goes: it must be possible, others do it too. I just find it a pity, I don't like to complain about Prusa.

In this light, one should use the findings for future decisions... e.g. chose components in designing new controller.. and this case really could be closed, unfortunately.

Dilbert

bkerler, 2024-09-01:
I just found an issue with ethernet handling the connection as half-duplex although full-duplex is possible. PR #4171 will resolve that issue. It won't help for wifi ... the wifi implementation is a uart-to-ethernet esp32 bridge, so speed is really limited. So at least for wifi, a different hardware would be required. Regarding ethernet I'm sure that there are still speed improvements left.

abjugard, 2024-09-06:
> I just found an issue with ethernet handling the connection as half-duplex although full-duplex is possible.

So just to clarify, what you mean is that the printers networking stack _thinks_ the connection is half-duplex, even though the switch and interface itself has properly negotiated a full-duplex connection?

> So at least for wifi, a different hardware would be required.

Again, for clarification, you mean a new xBuddy board design here right?

> Regarding ethernet I'm sure that there are still speed improvements left.

That is great to hear, looking forward to testing the changes once that PR has merged!

erwan, 2024-09-06:
Apparently the NFC card for the MK4S also does Wifi, does that mean it will be using this new hardware for wifi?

bkerler, 2024-09-07:
@abjugard Right now, the ethernet hw does feature duplex, but as a wrong hw register is being read, the connection is only set to half-duplex instead of duplex. This usually means that there are issues especially with huge transfers as the packets have to be resent many times. I could of course also have a big influence on transfer speeds, but right now I'm seeking for a way to test this.

bkerler, 2024-09-07:
@erwan some claim that with the mk4s wifi board, wifi speeds have improved. That could be the case if an esp32 is being used instead of an esp8266. But imho I wouldn't expect much. And it's not a nfc card but a nfc antenna that is just plugged into the existing buddy port.

Prusa-Support, 2024-10-03:
I seem to understand that full-duplex and NFC-antenna may not make a substantial difference but we will run further tests and make considerations. Thanks for your feedback.

Michele Moramarco
Prusa Research

[12 more replies]