prusa3d/Prusa-Firmware-Buddy #3050: [FEATURE REQUEST] Allow modification of extruder position for filament changes (other than top right corner)

BootsSiR, 2023-04-16:
Stock MK4

When loading or unloading filament, the extruder moves to the very top right corner of the build area to start the process.  This is problematic in enclosures with less top clearance as you're forced to work in an awkward position.

Please allow a setting that will move the extruder to the middle of the print area for filament changes.

MistrKernnunos, 2023-04-26:
You can set the position in the M600 gcode with X and Y parameter.

BootsSiR, 2023-04-26:
> You can set the position in the M600 gcode with X and Y parameter.

ok.  I may have been mistaken.. I think maybe I did a filament change when the printer didn't know it's x/y/z position so it went to the top right?  I've changed filaments since and it hasn't gone to that position

Prusa-Support, 2023-06-13:
Thanks for contacting us.

I believe the printer will always move up the Z-axis by a certain amount before parking, in fact, as emerged in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3022 it may touch the top when the Z-axis is specifically high before parking.
Some users want a bigger Z-lift https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2366, some users want a smaller Z-lift https://github.com/prusa3d/Prusa-Firmware/issues/3532. 

However, as pointed out by our collaborator - https://github.com/prusa3d/Prusa-Firmware/issues/3532#issuecomment-1359881651 -  in the near future it will be possible to customize a "Z-lift parameter" for certain gcodes.

I hope this clarifies.
The topic of the issue may look a bit confusing though.
Is there an ultimate request or feedback that you want to share?
Can this issue be closed?

Michele Moramarco
Prusa Research

BootsSiR, 2023-06-14:
@Prusa-Support I don't think this ticket is relevant any longer.  I haven't encountered this behavior again after reporting.

closed 2023-06-14 by BootsSiR