prusa3d/Prusa-Firmware-Buddy #3054: [BUG] Bring back support for long file names and timestamps in M20 L T g-code command

arekm, 2023-04-19:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - 4.5.4

**Original or Custom firmware** - Original

**Describe the bug**

M20 command issued over serial doesn't list long file names and doesn't list timestamps.
This feature existed for many years in MK3 firmware but is missing from MK4 firmware. 

**How to reproduce**

Issue
````
M20 L T
````
over serial.

````
Send: M20 L T
Recv: Begin file list
Recv: T-NUT_~1.GCO
Recv: End file list
````

**Expected behavior**

Long file names and timestamps (in form of hex numbers) are shown.

````
Send: M20 L T
Recv: Begin file list
Recv: T-NUT_~1.GCO 1061513 0x5690a164 "T-nut_M3_x12_20m.gcode"
Recv: End file list
````

**More info**
Code for that exists in MK3 firmware and also Marlin 2.1.2+ (guarded by LONG_FILENAME_HOST_SUPPORT and M20_TIMESTAMP_SUPPOR defs, so will be easier to see what's needed for backporting).

Prusa-Support, 2023-06-06:
Thanks for reporting.

Our developers will look into this.

Michele Moramarco
Prusa Research

bkerler, 2024-02-18:
I just wrote a PR #3765 for this issue.

danopernis, 2024-08-19:
This shouldn't have been closed since there are several PRs basically solving it.

timur-tabi, 2024-10-30:
Just is yet another feature that is missing in the MK4 firmware that was present in MK3.

timur-tabi, 2024-10-30:
> Thanks for reporting.
> 
> Our developers will look into this.

It's been almost a year and a half.  Maybe they should look harder?

redPanther, 2025-03-26:
please bring that feature. serial print is much more usefull with that!