prusa3d/Prusa-Firmware-Buddy #3056: [BUG] Filament runout sensor not working

skemp117, 2023-04-19:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - 4.5.4

**Original or Custom firmware** -Original

**Optional upgrades** - 

**USB drive

**Describe the bug**
A print failed because the end of the filament was not detected.

**How to reproduce**
Print something with not enough filament.

**Expected behavior**
Printer pauses when out of filament.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

reify3d, 2023-05-25:
I have managed to reproduce this error.

Printer type - MK4
Printer firmware version - 4.6.2
Original or Custom firmware - Original
Optional upgrades - None

Describe the bug
Filament sensor works intermittently. In the Sensor info>Filament sensor screen, with the idler door open, filament that is inserted should show "INS", while filament that isn't present should show "NINS". However, sometimes the filament sensor will show "INS" (false positive) when there is no filament present. The workaround is to calibrate the filament sensor, but some as-yet unknown cause will make this bug recur.

How to reproduce
Cut a piece of filament and let the filament sensor detect that filament has run out

Expected behavior
Printer pauses when out of filament

What actually happened
Printer continues to print without any filament getting extruded. Filament sensor has failed to detect that filament has run out.

G-code
Prusa Research-supplied Robo Alpaca on the Prusa Research-suppled USB stick is in Archive.zip

Crash dump file
In Archive.zip

Video
https://youtu.be/tuh1zAgMf9o

Workaround:
Perform a filament calibration. However, the root cause is not identified, and this problem may recur.
[Archive.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/11566655/Archive.zip)

Prusa-Support, 2023-06-14:
Hello! Thank you for the report and sorry for the late response on this thread. I noticed on the video that, in spite of the INS status being there all the time, the value next to it was changing (the changes from ~1,800,000 to ~30,000), I understood that was changing based on the filament inserting and pulling out, right? Or were these changes not consistent with it? 

In the meantime, we released newer firmware, with the 4.6.4 released very recently: can you please test it with the updated firmware, would the issue occur again? 

Let's keep an eye on the situation, in case of further issues even on the 4.6.4 firmware version, I'll relay feedback to our devs and we'll check it internally in even more detail. 

Alessandro Pantaleo
Prusa Research

Prusa-Support, 2023-09-08:
Is this issue still relevant?

Have you encountered this issue with the most recent FW 4.7.2 or 5.0.0-RC?
Also please keep an eye out for FW 5.0.0 stable version which will probably be released sometime soon.

In case you encounter this issue again though, please mind cleaning the filament sensor - it is surely worth a try.
- Re-inserting the filament a few times may help knock off debris from the filament sensor mechanism.
- Find the cleaning hole on the top of the Nextruder, in the middle of it, next to the brass ring, and blow air through it.
- If all of the above won't help, you can remove the "Nextruder sandwich" (PG brass ring altogether with the plastic parts, so you don't need to repeat the PG alignment) and remove the filament sensor for deeper inspection and cleaning.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-09-15:
This seemingly stale issue will be closed in favor of a duplicate that grew in popularity (https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3129).

Michele Moramarco
Prusa Research

closed 2023-09-15 by Prusa-Support