prusa3d/Prusa-Firmware-Buddy #3059: [ENHANCEMENT] Prusalink API should expose filament change required status

terba, 2023-04-20:
**Printer type** Prusa Mini+
**Optional upgrades** Filament Runout Sensor

**Is your suggested improvement related to an existing problem? Please describe.**
  It would be great, if PrusaLink and other software using the PrusaLink API could sense the filament change status of the printer. That would enable users to create automations for example in Home Assistant to send a message. Currently we have to watch a print when there is a possibility of filament run-out or there are color changes etc. This feature would be a huge comfort improvement with a little development cost.

**Describe the expected functionality**
  ```/api/printer``` endpoint should expose a new ```state.flags.filamentChangeRequired``` boolean attribute, and optionally this status of the printer could also be stated in the ```state.text``` attribute like _Filament change required._

terba, 2023-05-19:
Or in a more generalized way handling filament runout, M600 and M601 statuses with a ```state.flags.waitingForUser``` boolean attribute.

Prusa-Support, 2023-06-30:
Thanks for reaching out to us.

The lack of printer status/event notifications through Prusa Link is a known limitation at the moment.
The availability of further status/informative data will probably only be available via Prusa Connect in the near future.
Prusa Connect is not available for Prusa MINI at the moment (coming soon).

Please allow me to close this issue as mostly related to already existing issues (https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2822, https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2878, https://github.com/prusa3d/Prusa-Link-Web/issues/274 ... ).

Michele Moramarco
Prusa Research

closed 2023-06-30 by Prusa-Support

jhaand, 2024-04-27:
If Prusalink is the only locally supported interface, then Prusalink should also provide this status update. 

I was quite disappointed that I still have to check manually for color changes.