prusa3d/Prusa-Firmware-Buddy #3060: [FEATURE REQUEST] Calculating remaing filament on spool

wondalfo, 2023-04-21:
I would like to ask you, to include a general functionality in the printer firmwares. I came to that request, because I'am asking me quite often, if the remaining filament on the spool ist enough for the next print.
I have a Prusa Mini+, but the request is independent of printer type.

I would like to ask you, to implement several (10 should be enough) spool weight counters in the firmware. The following functions should be included:

- the actual weight of each spool shoud be stored, even with power off, maybe on the USB-stick
- at each filament exchange the printer asks, which spool is used or if a spool is replaced
- if the spool is replaced the printer ask for the amount of filament [in gramms] on the new spool
- before a print is started, the printer checks if the remaining filament on spool ist more than the calculated filament for this print
- if it is not enough, a warning should be displayed
- with every print the used amount of filament (this is already calculated by PrusaSlicer and displayed before print) is subtracted from the amount of filament on the spool
- before and after print the remaining filament of the used spool is displayed
- a inventory page, displaying all amounts of all filament counters would be helpfiull

additional functions

- the printer could even ask for the type and color of filament on spool
- this information should be displayed
- this information could also be used to preselct the temperatures for filament loading

Best regards Wolfgang

murk-sy, 2023-05-02:
Managing spools in the printer is not something I would do, this is more of a Prusa Connect type feature (especially for someone with multiple printers).

However, tracking filament usage on the current spool could be very useful just in general.

As a note: Printers only work with meters, rather than grams, which could mislead users that do not very accurately use the correct density setting in the slicer, especially for anything not very standard.

**As a fairly easy to implement middle ground, I propose the following:**
- Save current spool usage in meters separately of the extruder total 
- Display this in the statistics page along with approximate weight based on latest print
- Allow displaying usage in meters or grams on the footer

[image]

wondalfo, 2023-05-14:
Hi murk-sy, 

if you have only one counter, you loose all information in case you change filament, for example for printing in a different color. This is not very helpfull. 
I think more counters is not so difficult, and with exvery loading/exchange of filament the printer asks, which counter to use.

Best regards Wolfgang

murk-sy, 2023-05-15:
The problem with that approach is not multiple counters, but managing them. If you have 2 printers and swap filaments between them (or even 2 spools of the same color that you used at the same time), it becomes unusable without an external filament managing solution.

If you have red petg, neon red petg, orange red petg, black petg and cf petg, how will you keep track of them? Inputting names on the printer would be tedious, and using just numbers is probably not worth the effort, as it can get confusing fairly quickly.

My suggestion means that the physical spool is marked with the amount of used filament, ideally with a piece of paper. It is very low tech and requires some math if you load/unload it more than once (or just keeping track of the current amount).

Overall, I just don't think it's worth implementing a single-printer based filament management system. A lot of users who would benefit from that sort of thing probably have more than one printer - those would generally be prosumers and professionals - and most users with a single printer - most consumer users, that is - likely won't bother playing inventory management.

I see the benefit of inventory management, but I have 6 printers. Unless it's implemented in Prusa Connect, I just won't use it. It would be more expensive to manually track everything than to just have an extra spool in stock.

Sometimes I have specialty materials or colors I don't want to have in stock more than I absolutely need - and for those cases, just having the used amount listed on screen would be convenient, as I can just make a note on the filament spool itself.

Prusa-Support, 2023-06-16:
Thank you for your suggestion.

The logic behind seems quite simple and I like the idea, but the structure of a similar menu in the firmware may be more complex than what it looks like. I'm not entirely sure, but the memory in the Buddy board has certain limitations.
Moreover, managing data from several spools may lead to confusion, and it may be a bit difficult through such a basic LCD panel.

Our developers will make their considerations about this. Thank you again.

Until then, handling almost empty spools yet being aware of filament consumption can be simple enough.
- Equip the MINI with the filament sensor and forget it. The filament sensor will take care of the rest.
- The most attentive users may want to weigh the spool before use and check how much filament is needed according to Prusa Slicer before exporting the gcode. Prusa Slicer even offers an estimated minimum spool weight (filament + spool).
- The used filament information is briefed again in the print preview (before starting a print).
- Users can take advantage of the [placeholders](https://help.prusa3d.com/article/list-of-placeholders_205643) in the _output filename format_ field of Prusa Slicer to have print weight and filament length always stated in the file name together with the gcode essential information (e.g. `_{total_weight}g_{used_filament}m` ).

Michele Moramarco
Prusa Research

closed 2024-06-20 by github-actions[bot]