prusa3d/Prusa-Firmware-Buddy #3061: [FEATURE REQUEST] Preselect loaded filament in Preheat menu

Levi--G, 2023-04-23:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MINI] (or other buddy printers)
**Optional upgrades** - None

**Is your feature request related to an existing problem? Please describe.**
  At the moment the preheat menu always selects the top item while the loaded filament might be way more down in the menu. Flex for instance, which requires very long bed heating times, actually making preheat useful, is way at the bottom.

**Describe the expected functionality**
  I would love to see the menu preselect the loaded filament which is already stored somewhere for unloading. So i figure it will only be a minor change to preselect it in the menu, while keeping the other items where they are.
  This would mean:
  - you can double tap preheat to heat you current loaded filament rather than having to scroll the menu
  - You can still do other preheats, not impacting current functionality
  
  An additional Item that would make this even more useful would be detecting whether it is already preheating and then select cooldown instead, meaning you can swap between cooldown and preheating with consecutive double tapping :)
  
  Note: This might seem similar to #1795 but while that suggests a complete overhaul, this might be easier to implement sooner.
  
(I would implement this myself but since I don't want to break the pcb thing on my mini I will not attempt this sadly)

  Thank you :)

Prusa-Support, 2023-06-06:
Thank you for your suggestions.

Sounds like a quality-of-life improvement. Our developers will make their considerations about it.

On a side note, the [appendix](https://help.prusa3d.com/article/flashing-custom-firmware-mini_14) on the board is to protect inexperienced users from uncertified firmware versions that may promulgate in the wild internet.
Experienced users may consider breaking it at their own risk and virtually there are no other consequences once the warranty period is over (a matter of time).
We recommend caution when tweaking the firmware or flashing an unsigned version.

Michele Moramarco
Prusa Research

Levi--G, 2024-05-31:
This is still very much relevant.

CZDanol, 2024-07-30:
Internal ticket: BFW-5879

3d-gussner, 2025-09-25:
Please review your request with newest firmware.
There is the LCD option to manage filaments with Reorder and Enable filaments.

Levi--G, 2026-02-09:
If prusa team agrees its a good idea i can check if i can make it myself, but i will not invest time if they will deny the PR in the end.