prusa3d/Prusa-Firmware-Buddy #3068: [FEATURE REQUEST] Preheat Menu Enhancement

1John419, 2023-04-29:
Upon reflection, I would suggest an alternate solution for #3018 (and possibly #3019).

What if an intermediate menu were added to the Preheat menu, before the material selection menu? This menu would allow selection of the preheat type: nozzle, bed, or nozzle and bed. It would then proceed to the material selection menu and take the appropriate action base on preheat type selected.

Much easier than having to go Settings>Temperature>Nozzle and dial in the material temp when all you need is to heat the nozzle for cleaning or to load/unload filament.

murk-sy, 2023-05-02:
Duplicate/see also #1795

I would appreciate if you also share your thoughts on my suggestions there as well

Prusa-Support, 2023-06-08:
Thanks for reaching out to us.

However you suggest it, similar requests can be considered duplicates of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/706.
It may be of help if we keep all comments and suggestions together in a unique issue.

As an exception, we wouldn't mind keeping #1795 open due to its complexity and since there are a few extra points besides the notorious subject.
Usually, it's best to keep closely related comments altogether in a unique issue. Sometimes though, when the requests are many or rather specific, opening a separate issue for every single request may be more effective compared to a long list of requests in the same issue.

With that, I hope you don't mind me closing this issue as a duplicate.

Michele Moramarco
Prusa Research