prusa3d/Prusa-Firmware-Buddy #3070: [BUG] MK4 Power Panic Fails to resume print after extended brownout

reiuwu, 2023-05-07:
**Printer type** - MK4

**Printer firmware version** - 4.6.1

**Original or Custom firmware** - Original 

**Optional upgrades** - Stock

**USB drive or USB/Octoprint**
Provided USB

**Describe the bug**
Rapidly flickering power (car hit a transformer and we had a brownout lasting several seconds) causes the resume print function to fail. 

**How to reproduce**
Start a print. Disconnect AC power. Connect AC power. Quickly disconnect and reconnect AC power before the print resumes. Printer will boot to main menu

**Expected behavior**
 The printer should boot and resume the print. 

**G-code**
N/A

**Crash dump file**
N/A

**Video**
 N/A

Prusa-Support, 2023-06-10:
Thanks for reaching out to us and I'm sorry about the unfortunate happening, but this is not a bug.

The Power Panic can only recover from a sharp power outage.
This is a known limitation inherited from MK3 since the same good PSU is used on the MK4.
https://help.prusa3d.com/article/power-panic_2092
https://help.prusa3d.com/article/testing-safety-features-mk3-mk3s_2079

A special sensor detects mains voltage and in case of an interruption, it immediately shuts the power leaving enough power in the capacitors to store the strictly necessary data to recover from the power loss.
However the printer should be resilient to irrelevant power variations to prevent fake Power Panic triggering and, unfortunately, if the power loss is relatively gradual, the Power Panic won't be triggered promptly enough to spare sufficient energy for the capacitors to prepare a recovery.

This is not a bug but a known limitation, so please let me close this issue.

Michele Moramarco
Prusa Research

closed 2023-06-10 by Prusa-Support

rocrail, 2023-07-12:
After a short power loss the MK4 reboot sticks at 50%, so no recovery at all.
Firmware 4.7.0

Prusa-Support, 2023-10-03:
Not sure I understand but if you power off the recovery is not expected.
It can't be controlled either, it only occurs when the power outage is sudden.
If I got it correctly, the printer can't be able to recover from a reduction of power (brownout).

Michele Moramarco
Prusa Research