prusa3d/Prusa-Firmware-Buddy #3071: [BUG] (Mini) Transport of filament too short when unloading Filament

KlausKraemer, 2023-05-08:
MINI Fw 4.1.1 original

It's not often, but sometimes like today again: When unloading filament the transport time is too short and filament is still held fast in the extruder, because it didn't get tranported far enough back. Annoying ...

Simple to solve by extending the time while the extruder is running backward for e.g. 2 sec.

mix579, 2023-05-08:
Are you using the original PTFE Bowden tube? If it's a replacement and too long the filament won't be extracted properly.

KlausKraemer, 2023-06-01:
Sorry, vacation - I'm late ...

It's the original PTFE Bowden tube. I think manual movement of the extruder could be helpful anyway. My (still there, because of XL delays) Ultimaker has that and it's sometimes Helpful to be able to move filament "manually" like moving the axises.

Prusa-Support, 2023-08-01:
Thanks for reporting, however, we haven't heard of similar issues.

This doesn't sound like a bug but a hardware problem.

If the PTFE tube is original, then make sure the [extruder steps](https://help.prusa3d.com/article/experimental-menu-mini-mini_264396) value is set to _default_ (for original Prusa design and parts) or _tune it in case of a custom setup_.

Other possible causes of delayed unloading - off the top of my mind - may be:
- a bulky end of the filament (friction opposing the filament unloading),
- a stringy end of the filament (stringy end trapped inside the extruder after unloading),
- an incorrectly stopped [pinion](https://help.prusa3d.com/guide/3-x-axis-extruder-assembly_199519#200725) on the extruder motor (skipping steps),
- a loose [idler tension screw](https://help.prusa3d.com/article/idler-screw-tension_177367) (occasionally letting the filament slip),
- an [extruder gear dirty](https://help.prusa3d.com/guide/how-to-access-and-clean-the-extruder-pulley-mini-mini_126457) with debris or lubricant (letting the filament slip too).

Please let us know if this helped and consider reaching out to the [Customer Support](https://help.prusa3d.com/article/customer-support_2287) if the problem will persist.

Michele Moramarco
Prusa Research

KlausKraemer, 2023-08-02:
Thanks Michele,a bulky end of the filament is probably sometimes the problem.I had one last weekend, but the unloading routine kept asking for removing the filament from the filament sensor. I couldn't do that without using brute force but got stuck at that point in the FW. As there is no other way out I switched the Mini off and on again, reloaded the filament and let it purge. The next unload process went without a glitch.So it would be helpful if the Mini FW wouldn't insist on emptying the filament sensor to a need of switching off and on to get around it, but let one return to the filament menu.Have a nice evening Klaus

Prusa-Support, 2023-08-29:
Thank you for your suggestion.

To help form a neater shape at the end of the filament, from the firmware side, the printer would perform a slight purge before withdrawing the filament. Under normal circumstances this is sufficient but I may seem to understand that you would like to see some backup option alternative.
You may want to open a new specific issue about specific suggestions.

However, I'd like to underline that the bulky end of the filament is usually a symptom of improper temperature, assembly/hardware inconsistencies, or degraded/out-of-spec filament.

Please thoroughly review the [print-head assembly](https://help.prusa3d.com/guide/how-to-replace-a-heaterblock-heatbreak-mini-mini_122769) taking this chance to carefully check (or replace) the [PTFE tube](https://help.prusa3d.com/guide/how-to-trim-ptfe-tube-original-prusa-printers_22424#133076).
Also test and compare other freshly opened filaments, possibly one of those we provide pre-set profiles for. 
Last but not least, you can count on the help of our Customer Support.

Now that we have clarified the nature of the problem, can this issue be closed?

Michele Moramarco
Prusa Research

JonasKutzner, 2023-12-12:
Hi,
I also have this issue since switching to the new 5.1. FW. I never had an issue with properly unloading the filament before switching the Firmware. I basically hav it every time with PETG and PLA Filaments.

Prozewn, 2024-04-19:
Same for me... Happened after update, not sure which one. The filament stays stucked inside the extruder, If I pick that the unloading failed and pick try again option, it unloads it correctly, it's really annoying to unload or change the filament for 2 times...

closed 2024-07-08 by github-actions[bot]