prusa3d/Prusa-Firmware-Buddy #3072: [BUG] MK4 - print completes but watchdog reset error

BootsSiR, 2023-05-09:
Prusa MK4 running 4.6.1

Uploaded the print via PrusaConnect and started it on the printer.  I came back to my printer when the print should have been completed to find the print completed with this watchdog reset error message:

[image]

I don't know why it happened but on reboot the printer asked me to save a crash dump so here it is.

[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/11430785/dump_buddy.zip)

reiuwu, 2023-05-13:
I encountered the same issue at the end of a successful print. Attached BIN and GCode files. 

[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/11468538/dump_buddy.zip)

M4rc3lv, 2023-07-16:
In firmware version 4.7.0 this bug seems still to be there. See this forum message: https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-hardware-firmware-and-software-help/mk4-firmware-issue-watchdog-reset-immediately-after-finishing-a-print/

neuron-whisperer, 2023-07-24:
I'm the author of that post on Prusa 3D. Since submitting that post, I experienced a second instance of this error (with a different model).

In both cases, the error occurred within 30 seconds of finishing the print. I was watching the print status via both the web interface and a Nest camera (in anticipation of popping the print off the bed and starting a new one). The print finished and I walked from my office to the printer - about 50 feet - to find it displaying WATCHDOG RESET.

Since this appears to be an urgent issue, I spent a little time digging into it. As best I can tell, the firmware runs a watchdog thread that sends a heartbeat signal to the board every four seconds. Presumably, if the board doesn't receive it within a certain time, it raises an exception and disables output to the heating elements. It also appears that the watchdog thread performs some checks and can initiate its own exceptions, e.g., if it detects that a heating element is  activated when the firmware is not sending a signal, such as due to a logic error or a short.

It looks like this problem may have one of two causes:

1) The watchdog thread is being interrupted at the end of a print and fails to send a heartbeat signal within the watchdog window, causing the board to throw an exception.

2) The watchdog thread is throwing an exception at the end of a print upon detecting an anomalous condition (e.g., bed or extruder heatblock are too hot). I suspect that this is a threshold issue - e.g., the watchdog thread sometimes determines that the printer is idle but the bed or extruder are hot, and fails to recognize that that is the case _because the printer just finished printing._

I'll look at the crash dump to see if I can find more information.

voxelman, 2023-09-15:
This just happened after I used an SD card in a USB adapter to see if I could solve a very bad layer shift problem that I was having.
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12621651/dump_buddy.zip)

danopernis, 2024-03-21:
Please upgrade to latest firmware. Feel free to reopen this issue if the problem persists.

closed 2024-03-21 by danopernis