prusa3d/Prusa-Firmware-Buddy #3073: [FEATURE REQUEST] Allow cancelling a single object of a print instead of losing the whole print.

sylphio, 2023-05-09:
Prusa slicer is already Tagging all individual object during slicing.

When printing a large amount of objects on the Mk4 and even more on the XL , if a single object come of the bed during printer, if would be really useful to be able to cancel a single object instead of losing the whole print.

Octoprint have a similar feature that can be used as a reference and it was working well on the Mk3s.
https://plugins.octoprint.org/plugins/cancelobject/

Thank you

sylphio, 2023-05-09:
Similar to #481 but based on the object instead of a surface.

arekm, 2023-05-10:
There is a chance that this feature is already implemented in Prusa-Firmware-Buddy. PFB is based on Marlin 2 and it supports M486 command (https://marlinfw.org/docs/gcode/M486.html).

For this to work:
* slicer needs to be issuing M486 commands. PrusaSlicer doesn't do that(!). But SuperSlicer does.
* you need to know which object has what integer number assigned by slicer. No idea how to check that in SuperSlicer.

To test if this works you could:
* put simple/short model like 4 times on bed
* slice with superslicer
* verify that gcode file has M486 calls inside with different S values (object number)
* start print, let it finish 1 layer of all 4 objects
* while printing issue M486 C to cancel currently printing object or M486 P2 to cancel object number 2 for test
* check if printer started skipping printing that object

If this works then great - firmware supports M486 and only PrusaSlicer needs that support (https://github.com/prusa3d/PrusaSlicer/issues/3279)

sylphio, 2023-05-10:
@arekm Thank you for your answer. 

That would be great if at low level the firmware already support it. 
But most people don't know and/or are not setup to to send a M486 P### command to a running printing.
Some form of UI from the printer screen would be great.

P.s 
Prusa slicer already have support for object but it use comment instead of M486
[image]

arekm, 2023-05-10:
Yeah, that would be nicer since object names have some meaning for user.

Sadly that's not implemented in firmware (looked into Marlin 2 and PFB sources).

To send g-codes you need to connect to printer with usb cable and use any program with serial terminal capabilities (pronterface, putty, octoprint).

So half job is there, half is missing :/

sylphio, 2023-05-17:
Out of curiosity, it is even possible to connect to a running mk4? The board only have a single usb port and we need it for the storage device that contain the gcode. Prusalink maybe.

arekm, 2023-05-17:
Use usb-c that is on xbuddy board.

murk-sy, 2023-05-18:
Just as an idea, 2 quick on-printer GUI mockups

### A: Pregenerated image with numbers
[image]
[+] Entirely generated in slicer, relatively little work in firmware
[+] Only needs to display file list and a static image
[-] Numbers may be unreadable with a lot of small parts (or frankly in general)
[-] Requires scrolling through a file list and matching numbers

### B: X/dot marks the spot
[image]
[+] Better readability
[+] Does not actually require the file list displayed
[-] Requires reading model centers and more work in the firmware
[-] If a lot of identical parts are listed, finding the correct one may be tedious if they're not numbered in a logical manner

### Side notes:
- Prusa Mini (and I guess MK4 and XL I believe?) display is 240×320. Therefore you can fairly easily fit Mini and MK4 previews at 1px\~1mm and XL preview at 1px\~2mm. The above images are 160x160 px for the mini bed.
- Perhaps a combination of both would work best.
- [image]
Highlighting the entire object might be the fanciest way to do it that's easiest to read, but that would require a lot more image generation and is likely not worth the extra file size/effort.

Prusa-Support, 2023-06-22:
Thank you all for your input.

Our developers will look into this.
It goes without saying that this should work in synergy with PR https://github.com/prusa3d/PrusaSlicer/pull/3086 (reiterated by issue https://github.com/prusa3d/PrusaSlicer/issues/3279).

The feature was actually also requested and discussed on the "8-bit FW" branch some time ago (issue https://github.com/prusa3d/Prusa-Firmware/issues/2284).

Michele Moramarco
Prusa Research

murk-sy, 2023-10-20:
On-printer cancel object available in [5.1.0-alpha2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0-alpha2)

Prusa-Support, 2023-10-25:
Exactly.
The new firmware [5.1.0-alpha2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0-alpha2) allows this, however, it is still being tested and requires [PrusaSlicer 2.7.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-alpha1).

While these releases are not in the stable final version yet, the feature is officially implemented and functioning, so it seems safe to close this issue as solved.
Would you mind testing the feature and closing the issue then?

Michele Moramarco
Prusa Research

Prusa-Support, 2023-11-28:
The feature was fully implemented in the most recent firmware and slicer versions.

Michele Moramarco
Prusa Research

closed 2023-11-28 by Prusa-Support