prusa3d/Prusa-Firmware-Buddy #3074: [BUG] Preheat only goes to 170 °C nozzle temp

ulab, 2023-05-10:
**Printer type** - MK4 (Mini too)
**Printer firmware version** - 4.6.1
**Original or Custom firmware** - Original
**Optional upgrades** - 
**USB drive or USB/Octoprint** - irrelevant

**Describe the bug**
After selecting a preheat temperature in the menu, the nozzle temperature only heats up to 170°C, with the target temperature being shown at the correct value.

**How to reproduce**
Select `Preheat`, `PETG` and watch the temperature. It blinks red while heating to 170°C and then goes green around 170°C, showing 170/230°C

**Expected behavior**
I'd expect the nozzle to heat up to the chosen temperature.

BootsSiR, 2023-05-12:
@ulab 170C is the temperature at which the printer will do it's bed levelling routines.  It will then heat to the selected temperature prior to starting the print.  If you preheated to 230C and started a print, the printer would have to wait for the nozzle to cool to 170C before proceeding.

ulab, 2023-05-12:
Yes. But if I explicitely select to preheat to 230C, I want it to do that. Otherwise I'd chose to preheat to 170C.

I may want to push some cleaning filament through the hotend, hot-tighten the nozzle or something else and not turn the knob forever in the Settings menu.

If Preheat let's me chose 230C and then just heats up to 170C, it's irritating. Especially if it shows 170/230C, but just doesn't heat up anymore even though the target temperature of 230C has not been reached.

BootsSiR, 2023-05-12:
In your case I'd just set the nozzle temperature to 230C in the control menu.  Preheat seems to be intended to preheat for a print.

murk-sy, 2023-05-15:
It's not really a bug, but it is definitely confusing if you don't already know exactly what the function does.

I've made a feature request #1795 that should cover this (since it adds a function + clarifies what it does). I'd appreciate your thoughts there.

Prusa-Support, 2023-06-23:
Thanks for writing.

Please select `filament` -> `load`/`unload`/`change`/`purge`.
Alternatively, `settings`-> `temperature` if you have reason to skip the operations of the motor _while heating up_.

The _preheat_ option is actually intended for the preparation of the printer for the print, which would imply noticeable downtime if the nozzle was always reaching the target only to cool down before the _mesh bed leveling_.
All in all, this is not a bug; on the contrary, this is intentional.

Could the issue be closed?

Michele Moramarco
Prusa Research

Prusa-Support, 2023-07-14:
Thank you again for your input and I hope you are satisfied with the answer.
It seems safe to close this issue now.

Michele Moramarco
Prusa Research

closed 2023-07-14 by Prusa-Support