prusa3d/Prusa-Firmware-Buddy #3075: [ENHANCEMENT] Ask before changing filament manually

Gyrxiur, 2023-05-13:
**Printer type** - [MINI]
**Optional upgrades** - [Filament Runout Sensor]

**Is your suggested improvement related to an existing problem? Please describe.**
My rotary encoder sometimes jumps to next item, when I click on it, and when changing speed during print, sometimes it jums to change filament and that gives me headaches.

**Describe the expected functionality**
It would be great, if printer asks user if he really wants to do it, before manually changing filament or unloading it, during or not during printing, just if his rotary encoder is not working perfectly.

Gyrxiur, 2023-05-13:
Or make it a setting, if user wants to be asked before doing anything with filament.

Prusa-Support, 2023-06-16:
Thank you for your suggestion.

This looks like a quite particular hardware problem, maybe negligible, but it doesn't seem to be a common issue.
I don't want that a not-so-sharp movement of the knob leads you to overcompensate the rotation. Possibly remove the printed knob and clean the gap down the encoder with canned air and floss.
Consider consulting [our official Support](https://help.prusa3d.com/article/customer-support_2287) if the problem occurs frequently.

When the knob movements are sharp, it is unlikely to select an undesired option in the menu.
However if the knob issue will be reported multiple times or the feature request becomes popular, we will definitely take this into consideration. 

Michele Moramarco
Prusa Research

Prusa-Support, 2023-06-16:
On a side note, a minor "frozen knob input" bug was detected by our developers and fixed in [FW 4.0.5](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.0.5).
I'm assuming your firmware is up to date. If not, please upgrade the firmware.

Michele Moramarco
Prusa Research

closed 2024-06-20 by github-actions[bot]