prusa3d/Prusa-Firmware-Buddy #3076: [BUG] MK4 Nozzle Temperature only goes to 290C

dp250f, 2023-05-19:
**Printer type** - [MK4]

**Printer firmware version** - [4.6.2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Copper/Hardened Steel tip Nozzle with Nextruder V6 Nozzle adapter]

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/PrusaLink

**Describe the bug**
  Printer specs state "Max nozzle temperature 300°C" but it only goes to 290°C

**How to reproduce**
  Try to set nozzle temperature to 300 - it stops at 290

**Expected behavior**
  Nozzle temperature goes up to 300

**G-code**
  N/A

**Crash dump file**
  N/A

**Video**
  N/A

dp250f, 2023-05-28:
Still the same in 5.0.0-alpha1

dp250f, 2023-06-23:
Still not fixed in 5.0.0-alpha2

dp250f, 2023-07-07:
Still a problem in 5.0.0-alpha3

dp250f, 2023-07-29:
Still a problem in 5.0.0-alpha4

dp250f, 2023-08-14:
Still a problem in 5.0.0-RC

girak49, 2024-01-09:
The [specs](https://www.prusa3d.com/product/original-prusa-mk4-2/#specs) now state the max nozzle temp is 290C.
Anyone know why we lost 10 degrees?

closed 2024-06-20 by github-actions[bot]