prusa3d/Prusa-Firmware-Buddy #3077: Sending (post) printer status (JSON)

VladimirVecera, 2023-05-19:
I would like to check the status of the printer when away from home. Would it be possible to implement a POST method in the firmware for sending, for example, JSON? POST due to the fact that not everyone has a fixed IP address and could subsequently process the data on any web hosting.

My printer: Mini+

murk-sy, 2023-05-20:
Prusa Connect support is planned *at some point*, so the printer sending some form of active updates will be added. I'm not sure you can change to a different service by default (as in, just change the domain to which you're sending data), but in worst case you should be able to do it by modifying the firmware.

Prusa-Support, 2023-07-04:
That's correct. Prusa Connect will enable full remote control on the printer when Prusa MINI will be enabled on it (coming soon).

Michele Moramarco
Prusa Research

Prusa-Support, 2023-11-01:
The new firmware version (now available as alpha - potentially unstable) enables Prusa MINI printers on Prusa Connect, which allows the user to know the status of the printer, setting up [notification systems](https://help.prusa3d.com/search?s=connect+notifications+setup), and much more.

I hope you don't mind me closing this issue now.

Michele Moramarco
Prusa Research

closed 2023-11-01 by Prusa-Support