prusa3d/Prusa-Firmware-Buddy #3080: [BUG] 4.4.1 on Prusa Mini will not reconnect to octoprint after a disconect with resetting printer

Sp4rkR4t, 2023-05-25:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI

**Printer firmware version** - 4.4.1

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint** - Octoprint

**Describe the bug**
About 50% of the time when octoprint has crashed or I change something on the Pi needing a restart when octoprint comes back up it cannot negotiate a connection with the printer unless I reset the printer as well.

**How to reproduce**
1/ Connect octoprint to Mini.
2/ Reset machine running octoprint
3/ try to reconnect to mini, roughtly 50% of the time it will fail until printer is reset.

**Expected behavior**
Printer should be able to re-establish connection with octoprint.

Prusa-Support, 2023-07-05:
Thank you for reporting.
On a side note, I think this may be a known issue that was reported before, but I can't confirm this as I was not able to spot similar or directly related issues.

Michele Moramarco
Prusa Research

marek26340, 2023-08-03:
Hi, I suggest you to just follow the official docs to downgrade the firmware on your MINI back down to 4.3.4.
All of the new firmware versions (4.4 and above) have proved to be very buggy for me (#3042 #3036).
Don't bother with any newer versions after 4.3.4 until the devs fix it up.

Crowlord, 2023-08-16:
Another day another octoprint breaking bug. 

Its OK they will deny its an issue soon and blame octoprint.

Dunnymeister, 2024-05-27:
I have not experienced this issue lately - perhaps it is ok to close?

closed 2024-08-27 by github-actions[bot]