prusa3d/Prusa-Firmware-Buddy #3081: [ENHANCEMENT] Prompt to eject filament on M600 change

kozross, 2023-05-25:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MINI
**Optional upgrades** - 

**Is your suggested improvement related to an existing problem? Please describe.**
Whenever I do a filament switch via M600 (such as when I use PrusaSlicer to set up a colour change), the ejection happens without a prompt. This is quite annoying (and doesn't match the behaviour of the MK3S+), as it means I have to carefully babysit my print to avoid spool tangles.

**Describe the expected functionality**
A prompt should ask before ejecting.

Prusa-Support, 2023-07-07:
Thanks for your feedback.

In the specific case of Prusa MINI, the filament inlet tubing will prevent the filament end from slipping outside and creating a tangle. Possibly don't operate without the [PTFE tube 4x2x150 mm](https://help.prusa3d.com/guide/3-x-axis-extruder-assembly_199519#201222) inlet tubing.

A tangle can't happen as long as the filament is held in the tube.
In the worst scenario, a coil around the spool will temporarily become slack. The filament with be wound back snugly around the spool after a little rotation.
As usual, hold the end of the filament all the time during the filament handling, wind it your best, and secure the end of the filament on the spool.

Other than this, we appreciate your suggestion.

Michele Moramarco
Prusa Research

kozross, 2023-07-07:
I have always used the tubing you've mentioned. I've had tangles before, several times. It's not filament-specific either: I've had this problem with several kinds of filament.

closed 2024-06-20 by github-actions[bot]