prusa3d/Prusa-Firmware-Buddy #3082: [BUG] Toolhead freezes after filament changes

kozross, 2023-05-26:
P**Printer type** - MINI

**Printer firmware version** - 4.4.1

**Original or Custom firmware** - Original

**Optional upgrades** - 

**USB drive or USB/Octoprint**
  USB drive

**Describe the bug**

After M600 filament changes, I periodically get toolhead freezes. The toolhead stops moving and just sits there. The only fix is to pull out the USB drive, plug it back it, then resume. Resuming without pulling out the USB and re-inserting just causes the toolhead to park, then unpark and stay put again.

**How to reproduce**

1. Slice any print with an M600
2. Have the M600 trigger, switch filaments, restart print.
3. Watch the toolhead occasionally stop mid-print.

**Expected behavior**

The toolhead shouldn't stop like that, or if it does, it should specify why exactly this happened.

**G-code**

[Attached as zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/11570313/vertex-rest-tiles-2_0.2mm_ABS_MINI_3h48m.zip). This is my most recent example that triggered this.

**Crash dump file**

None to provide.

**Video**

None to provide.

Prusa-Support, 2023-07-07:
Thanks for reaching out to us.

I don't seem able to reproduce this.
Moreover, I don't seem to see similar reports despite FW 4.4.1 being around since January.

Regardless of this, possibly test again after _downgrading the firmware_ (4.4.0 or older).
Also, please try again with a _different USB flash drive_.

Is the printer equipped with a filament sensor?
Please confirm there no is extra hardware connected to the printer (Wi-Fi ESP module, Ethernet, devices connected via micro-USB cable...).

Michele Moramarco
Prusa Research

Prusa-Support, 2023-08-01:
So far there haven't been similar reports and we couldn't confirm this issue.
Possibly provided more information and evidence like a video.
If we can't confirm the problem, the issue may get closed as stale.

Michele Moramarco
Prusa Research

kozross, 2023-08-01:
@Prusa-Support - feel free to close. I'm giving up on the Mini completely, in light of this, and _many other_, issues, many of which go back years, and haven't even been acknowledged, much less worked on. I am _super_ disappointed with this product, as well as Prusa as a company in general, given this, and recent directions you seem to be taking.

Prusa-Support, 2023-08-25:
I'm sorry to hear that but I'm afraid this can be much of help to clarify and possibly resolve the problem.
I'll close the issue at this point and I really hope you don't mind.

We remain available to help via the official Customer Support channels and we will keep collecting feedback and reports.

Michele Moramarco
Prusa Research

closed 2023-08-25 by Prusa-Support