prusa3d/Prusa-Firmware-Buddy #3084: [BUG] Cannot change filament when print is paused

kozross, 2023-05-26:
**Printer type** - [MINI]

**Printer firmware version** - 4.4.1

**Original or Custom firmware** - Original

**Optional upgrades** 

**USB drive or USB/Octoprint**

USB drive

**Describe the bug**

When the MINI+ is paused mid-print, you cannot access the 'Tune' menu to change filament. This was quite surprising to me, and is poor UI. I had to speak to a support staffer to figure out that in fact, you should use 'Tune -> Change Filament' mid-print, which I would never have guessed.

**How to reproduce**

1. Start any print.
2. Pause it.
3. Try to access the 'Tune' menu.
4. Note that you can't.
5. Lament.

**Expected behavior**

Changing filament while paused should be possible.

**G-code**

Not applicable.

**Crash dump file**

Not applicable.

**Video**

None taken.

Prusa-Support, 2023-07-08:
Thank you for reporting.

I believe this is a known issue indeed - not sure, I can't find any documentation about it.
Thanks for the accurate 5 reproducibility steps. 🙂 

Our developers will look into this.

Michele Moramarco
Prusa Research

Prusa-Support, 2024-01-27:
Correct. Because the base code of the firmware is substantially different now - now that it is consistent with Prusa MK4 and Prusa XL - the problem shouldn't persist and several other issues based on FW 4.x.x are potentially obsolete now.

I would close this issue if you don't mind.

Michele Moramarco
Prusa Research