prusa3d/Prusa-Firmware-Buddy #3086: [FEATURE REQUEST] Remeaning time instead of end print time

sybaris, 2023-05-27:
Hi,

**Printer type** : [MINI]

Is it possible to have a settings to display remaning printing time instead of end print time ?
The problem of end print time : 
1°/ You must have a clock or a watch to know the current time, to appreciate when it will be finish
2°/ If the printer si not at the correct time (timezone pb or ntp pb)...

Thanks for advance

Regards
Sybaris

cb88, 2023-06-01:
Time remaining and current layer for example are more useful to me than the existing displayed info. 

Time remaining is actually what shows by default if it isn't networked.

reedy, 2023-06-06:
Dupe of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/648

Prusa-Support, 2023-08-25:
Sorry, this issue should be closed as a duplicate.

Michele Moramarco
Prusa Research

closed 2023-08-25 by Prusa-Support