prusa3d/Prusa-Firmware-Buddy #3087: [FEATURE REQUEST] Prusa link : display files sorted by date

sybaris, 2023-05-27:
Hi

**Printer type** - [MINI]
Currently, when I watch my files on Prusalink, they are displayed in alphabetical order.

Is it possible to add a combobox on this web page to choose a sort by date order?

Thanks for advance
Regards
Sybaris

dimitarsp, 2023-05-28:
+1 for this. I am so used to order when plugging in the USB directly to the printer, that it is feels really odd not to have it the same way in prusa link

Prusa-Support, 2023-08-01:
Thanks for your feedback.

This is actually a duplicate of the issue https://github.com/prusa3d/Prusa-Link-Web/issues/309.
It should be already solved on the Prusa Link's side, but the feature will be made compatible with Prusa MINI in a future printer firmware upgrade.

Michele Moramarco
Prusa Research

ChrisHeerschap, 2024-02-11:
How future? I've got the current 5.1.2 firmware on my Mini and it's still sorting alphabetically, and it's been over six months since the last comment.

tkircher, 2024-02-24:
Same issue with the MK4, no ability to sort by date. I only ever want to see the most recent upload at the top of the list, I have no interest in alphabetic sorting.

Prusa-Support, 2024-04-04:
For now, as far as I can tell, the feature is available on the core Prusa Link system which can be flashed on Raspberry boards.
In fact, an external Raspberry board offers high resources and a complete operating system running on its own.

Michele Moramarco
Prusa Research

tkircher, 2024-04-04:
> For now, as far as I can tell, the feature is available on the core Prusa Link system which can be flashed on Raspberry boards. In fact, an external Raspberry board offers high resources and a complete operating system running on its own.
> 
> Michele Moramarco Prusa Research

Let us all know how we can set up Prusa Link on a Pi for controlling a MK4.

ChrisHeerschap, 2024-04-04:
> For now, as far as I can tell, the feature is available on the core Prusa Link system which can be flashed on Raspberry boards. In fact, an external Raspberry board offers high resources and a complete operating system running on its own.

That might also fix the slow wifi upload issue, but let's be honest, if I'm setting up a raspi to connect to my MK4, I'm running Octoprint because it's still got way more functions and features and plugins.

reinux, 2024-08-02:
This is a surprising omission. How often does anyone need to find something by date as opposed to by the most recent items?

The UI element is already there for selecting other sort criteria, which kind of makes me wonder if maybe the system calls for digging into file metadata beyond just the filename is prohibitively expensive or difficult or something. I've seen that be an issue even on desktop OSes under certain extreme conditions.

Could you please let us know if that's the case? If we shouldn't expect a solution near-term, no problem, I'll just install Octoprint.

lsr1, 2024-08-17:
+1
Came here to support this Feature Request.

tkircher, 2024-08-17:
Still waiting for this, or for full Octoprint support, neither of which is available yet.

tkircher, 2024-09-18:
Octoprint support is still only at around 80%, but I switched over completely and haven't looked back. Prusa link is a disaster and can't be fixed.

3Dabbler, 2024-12-17:
+1 for PrusaXL.  Latest firmware, I don't see the ability to sort by date.  Makes it really hard to print the file I just uploaded (buried by alphabet down somewhere).

PhilipWhiteside, 2024-12-30:
I made a browser script to do this for the Mini https://gist.github.com/PhilipWhiteside/8a60bcc6d48a91d50b1bae9e7120f8fa.

sybaris, 2025-01-03:
Hi,

Thanks for this very good idea.
For my part, I have just to adapt a little the script because my date format is in french.

Here the change for french dates :

```
  const parseDate = (dateString) => {
      const [day, month, yearAndTime] = dateString.split('/');
      const [year, time] = yearAndTime.split(' ');
      return new Date(`${year}-${month}-${day}T${time}`);
  };
```

I didn't know ViolentMonkey. I think I will use it.
Thanks

Regards
Sybaris

danopernis, 2025-01-14:
Hi folks, this is scheduled for 6.3.0 Sorry for taking so long, this needed coordination from more people than I would have liked.

Guidhouse, 2025-03-02:
> Hi folks, this is scheduled for 6.3.0 Sorry for taking so long, this needed coordination from more people than I would have liked.

I can see the 6.3.0 is released for the Core. 
When can we expect XL and others to follow?

CZDanol, 2025-03-03:
Sorry, we don't have any estimates on that yet.

reinux, 2026-01-19:
Errr... bump? I guess?