prusa3d/Prusa-Firmware-Buddy #3088: Mesh bed leveling 7x7 (MINI+)

VladimirVecera, 2023-05-28:
I couldn't find a better mesh leveling setting anywhere in the printer settings. I wanted to set the printer to 7x7 dots, but I can't see it anywhere in the settings. Is there a possibility to have a better mesh bed leveling setting for the MINI+ printer as well?

mix579, 2023-05-28:
There's no 7x7 MBL on the Mini, only the Mk3S. The Mini only has 4 x 4 due to the smaller bed size. 

You can close this issue.

murk-sy, 2023-06-08:
Duplicate of #1264 
There's also Prusa Support's reply there regarding this

Prusa-Support, 2023-06-16:
Hello, thank you for the request and thank you @murk-sy for pointing to the duplicate issue with our response - closing this particular thread being it a duplicate, but feel free to add any comments to the original thread #1264 

Alessandro Pantaleo
Prusa Research

closed 2023-06-16 by Prusa-Support