prusa3d/Prusa-Firmware-Buddy #3090: Y axis calibration fail on update[BUG]

derekgdicamillo, 2023-05-28:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - mk4

**Printer firmware version** - alpha with input shaping

**Original or Custom firmware** - [Original/Custom]

**Optional upgrades** - none

**USB drive 

**Describe the bug**
  After installing alpha firmware y axis will not calibrate. It continues to fail. 

**How to reproduce**
  Please describe steps to reproduce the behavior.

**Expected behavior**
  A clear and concise description of what you expected to happen.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

Prusa-Support, 2023-08-02:
Thanks for reaching out to us.

We can't confirm the presence of a bug.
This may commonly happen on any firmware version for various reasons related to the axis motion, one above all the belt tension. How cleaned and lubricated are the rod may surely affect the amount of friction over the axis movements, but we can't exclude minor secondary settling and modification of the geometry over the initial use.

I was under the impression that a slight belt-tightening is a frequent solution, however, please try both tightening and loosening the belt tension screws by +/- one turn to see if this makes a difference, and feel free to report back.

May the problem persist, please refer to the [Customer Support](https://help.prusa3d.com/article/customer-support_2287).

Michele Moramarco
Prusa Researchh

Prusa-Support, 2023-08-24:
We can't confirm a bug and we haven't got similar reports recently.
Also, we haven't heard back from you so I hope you got this sorted with the help of our Customer Support.

This issue will be closed as stale now but feel free to submit further feedback below.

Michele Moramarco
Prusa Research

closed 2023-08-24 by Prusa-Support