prusa3d/Prusa-Firmware-Buddy #3091: [BUG] Crash while printing a file that I sliced myself with the new profiles

wlayher, 2023-05-29:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK4IS]

**Printer firmware version** - [5.0.0-alpha1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
  Printing via OctoPrint

**Describe the bug**
  The print was running and for about half an hour and then the printer crashed with a red screen on the display "Input_shaper_state" "Unexpected state 1"
[image]

**How to reproduce**
  Printing the file via Octoprint

**Expected behavior**
  I expected the file to print successfully, as it was not the first file I printed the same way.

**G-code**
  
[3D4U_Coffee_Clip_Large_V3.1_HC_with_Cap_MK4IS_0.4n_0.2mm_PLA_MK4IS_4h7m.aw.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/11591474/3D4U_Coffee_Clip_Large_V3.1_HC_with_Cap_MK4IS_0.4n_0.2mm_PLA_MK4IS_4h7m.aw.zip)

**Crash dump file**
  
[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/11591476/dump.zip)

**Video**
  I have no video of the exact time of the crash.

Prusa-Support, 2023-08-03:
Thanks for reporting.

We haven't come across similar reports so far so we can't exclude a temporary glitch.
We should work on the reproducibility of this issue.

Have you been able to reproduce this issue while printing this same gcode later again?
Would it make a difference if the same file is directly loaded on the USB flash drive avoiding any network data transfer?
Would the issue ever occur while printing any other print file?
Would the issue persist if the gcode is not post-processed with ArcWelder?
Would the issue ever occur with Octoprint's board totally disconnected?

Can you reproduce the issue on FW 4.7.2 (very new) or 5.0.0-alpha4?

Michele Moramarco
Prusa Research

wlayher, 2023-08-03:
Hello Michele,
it has been some time since my report, but I'll try to remember.
> Thanks for reporting.
> 
> We haven't come across similar reports so far so we can't exclude a temporary glitch. We should work on the reproducibility of this issue.
> 
> Have you been able to reproduce this issue while printing this same gcode later again? 
I haave not tried to print it in exactly the same way again so far.
>Would it make a difference if the same file is directly loaded on the USB flash drive avoiding any network data transfer?
I printed the file successfully loading it directly from the USB flash drive, same firmware as above. The error occurred again a few times with other files when I printed gcode with input shaping via OctoPrint and the success rate was better when I printed those from the USB flash drive.
> Would the issue ever occur while printing any other print file? 
I remember that the same error occurred again with at least another print file but I did not keep records for that.
> Would the issue persist if the gcode is not post-processed with ArcWelder? 
As far as I remember it does not matter if the file was processed with ArcWelder or not. I have printed gcode with input shaping successfully from the USB flash drive that were post-processed with ArcWelder.
> Would the issue ever occur with Octoprint's board totally disconnected?
I switched to printing IS gcode via the USB flash drive. The success rate is higher with this method. I am not sure if the same error reoccurred when printing from USB flash drive.
> 
> Can you reproduce the issue on FW 4.7.2 (very new) or 5.0.0-alpha4?
I have not tried the same file with other firmware versions so far.  I have tried all firmware versions up to 4.7.1 and 5.0.0-alpha4 with other gcode with input shaping but I did not keep records for failed prints. A few times the Tasmota plugin from OctoPrint shut off the printer when it lost the connection before I could see the error screen so I cannot be sure if that was the same error or the error described below.
During the hot weather 3 weeks ago I ran into another error, a red error screen with "freertos - configASSERT" that occurred when the printer had very high temperatures, ambient in the room with the enclosure as well as on the board. I am in the process of trying to implement some way of cooling for the board and until this is finished I have not used the printer for about two weeks. As soon as I can use the printer again I will try printing the same file again with the same conditions to see if I can reproduce the error with newer firmware.

Kind regards,

Walter

Prusa-Support, 2023-08-30:
We haven't been able to reproduce the error and I seem to understand that you didn't come across the problem again in recent times. Also, we apparently haven't come across similar reports.
In case of hardly reproducible glitches please consider performing a [factory reset](https://help.prusa3d.com/article/factory-reset-mk4-xl_421206) as the first step.

All in all the issue may not be necessarily related to Input Shaper features and profiles either.
If sending the gcodes OTA or uploading them directly on the USB flash driver makes a difference, then this may be related to a different issue (https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3156).

Finally, this seems to be a temporary glitch with an unstable Alpha firmware, which is obsolete by now.
Please test the [most recent firmware version](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases).

It seems safe to say that the issue is no longer relevant or obsolete.
Would you mind this issue?

Michele Moramarco
Prusa Research

wlayher, 2023-08-30:
I will try to print the file again with the most recent firmware during the next few days. I installed a fan for active cooling on the xBuddy case, see [here](https://forum.prusa3d.com/forum/postid/671000/). I am testing this now. Since the beginning of my tests I did not encounter the problem again. As of now I did only test with PLA to establish a base line, I will test with PETG, ABS, ASA during the next days.

wlayher, 2023-08-31:
I just printed the same gcode file as in the attachment of the first post again, successfully. Firmware 5.0.0-rc, printed via OctoPrint. After installing the fan that I mentioned in the previous post I have printed nonstop with the printer for about 5 days without any issue, all prints were gcode files created with IS and printed via OctoPrint and processed with the Arc-Welder plugin on OctoPrint. So I think it is safe to assume that the problem was overheating on the xbuddy board.

closed 2023-08-31 by wlayher