prusa3d/Prusa-Firmware-Buddy #3093: [FEATURE REQUEST] Move nozzle to center of head bed while cleaning

stritti, 2023-06-03:
**Printer type** - [MINI]

**Is your feature request related to an existing problem? Please describe.**
  If I switch filament on my Prusa MINI+ the nozzle is moved to the left top edge of the heat bed.

  When the filament is extruded to get the new color clean, the filament drops often behind the head bed under the printer.

**Describe the expected functionality**
  
  Would be nice if the nozzle would be moved some centimeters more to center while it is heated. Then the filament would be extruded on the head bed and can be removed more easily.

murk-sy, 2023-06-05:
Side note: this position used to be X~40 but was changed to X0 to fix #2323 
X0 should still be used if filament change is used during a print

Prusa-Support, 2023-08-24:
Thanks for your suggestion.
Our developers will consider this.

Michele Moramarco
Prusa Research

stritti, 2024-05-30:
Would be really nice not to purge behind the print bed. Currently I move every time the bed manually.

closed 2024-10-08 by github-actions[bot]