prusa3d/Prusa-Firmware-Buddy #3094: M150 is not enabled / implemented to control LED on the XL

droomurray, 2023-06-04:
M150 is not completely implemented / enabled in the firmware to control the LED's

Be great if it was as it looks partially coded ?

Prusa-Support, 2023-08-04:
Thanks for reporting.

Our developers will look into this.
For now, please notice the user has partial control over the lights via `Settings > User Interface`.

Michele Moramarco
Prusa Research

Harms3D, 2023-11-11:
Hello Prusa Support,

I can't control the LEDs on the MK4 via the RepetierServer. I have already read various M150 documentations and have come to the conclusion that M150 is either deactivated or used differently than the documentation says. Apparently I am not the only 

one who has this problem and I would be happy if this nice feature of the MK4 could be used. Please help.
https://forum.prusa3d.com/forum/hilfe-zur-hardware-firmware-und-software-3/mk4-status-led/#post-683629

Harms3D, 2023-11-11:
Maybe you don't know it, but if you run an MK4 via a repeater server, the LEDs are unfortunately off. They only turn green once when booting and then there are no more actions. However, the LED events could be controlled via the repeater server if M150 worked.
The Support from Repetierserver cant help here, because its a Problem with the MK4:
https://forum.repetier.com/discussion/comment/45124#Comment_45124

anghelos, 2024-01-22:
This is true for the MK4 too.
The official [documentation on status LEDs](https://help.prusa3d.com/article/status-led-explained-mk4-xl_418975) says that they can be controlled via M150 command, but that doesn't seem to be the case. So this would make this more of a bug than an enhancement?

Prusa-Support, 2024-01-24:
The code `M150` is listed among the currently compatible codes in the [Buddy firmware-specific G-code commands](https://help.prusa3d.com/article/buddy-firmware-specific-g-code-commands_633112) Help article, so I would expect it to work. 
The parameters in our article are slightly different than the ones explained in [marlinfw.org](https://marlinfw.org/docs/gcode/M150.html) though.

I will have the developers and the content team to double-check again the functionality of this code and the articles

Michele Moramarco
Prusa Research

anghelos, 2024-01-24:
> The code `M150` is listed among the currently compatible codes in the [Buddy firmware-specific G-code commands](https://help.prusa3d.com/article/buddy-firmware-specific-g-code-commands_633112) Help article

This is the one I followed to do some tests (with HSV), with no luck.
It would be also great to have a bit more information on what values the A (Animation type) and S (printer status) parameters accept.

Prusa-Support, 2024-02-03:
Yes, the lack of `A` and `S` parameters was immediately reported and the content team took action.

> A: animation type
(SolidColor / Fading)

> S: printer state
>        -   Idle,
>        -   Printing,
>        -   Pausing,
>        -   Resuming,
>        -   Aborting,
>        -   Finishing,
>        -   Warning,
>        -   PowerPanic,
>        -   PowerUp

       
The article has been updated.
https://help.prusa3d.com/article/buddy-firmware-specific-g-code-commands_633112

However, at the moment, the code appears not to be fully functional as intended.

Michele Moramarco
Prusa Research

closed 2025-05-12 by github-actions[bot]